open62541/open62541 #1472: client: local TCP connection. Preferable interface type (IPv4, IPv6)

ptvoinfo, 2017-12-22:
--------------------------------------------------------------------------------

Description
===========

A client establishes connection to a server in local network using a network name. The server listens on an IPv4 address only.

The function "UA_ClientConnectionTCP" in "ua_network_tcp.c" tries to resolve a host name to an IP address using:

int error = getaddrinfo(hostname, portStr, &hints, &server);

This function returns a list of network interfaces. In some cases the first interface in the returned list is IPv6. The client does not allow to select a network interface. 

I think that is better to use IPv4, especially for local network connections or try to connect twice (or more times) on all interfaces.

Checklist
=========
 - [ ] open62541 Version: master / Dec 20, 2017
 - [ ] Operating system: Windows 10

jpfr, 2017-12-23:
getaddrinfo returns IPV6 first according to the standard.
https://tools.ietf.org/html/rfc6724

You can change this behaviour on Linux via the /etc/gai.conf file.

Looping over the different interfaces until a connection works is a good idea!
Any code contributions for this are welcome.

HolgerJeromin, 2017-12-25:
Reference: https://en.m.wikipedia.org/wiki/Happy_Eyeballs

ptvoinfo, 2017-12-27:
In my OPC client I know a target OPC server. I know this OPC server supports IPv4 only. This is not a best solution try to connect to IPv6 first (because it requires additional time and depends on 
I think it would be great to allow to select an interface type (IPv4, IPv6 or auto) through UA_ClientConfig when I'm creating a new client connection.