open62541/open62541 #1473: Build errors on FreeBSD: error: use of undeclared identifier 'INFINITY', 'NAN', etc

yurivict, 2017-12-23:
```
src/ua_types_encoding_binary.c:384:34: error: implicit declaration of function 'signbit' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    else if(f == 0.0f) encoded = signbit(f) ? FLOAT_NEG_ZERO : 0;
                                 ^
src/ua_types_encoding_binary.c:384:34: note: did you mean '__signbit'?
/usr/include/math.h:192:5: note: '__signbit' declared here
int     __signbit(double) __pure2;
        ^
src/ua_types_encoding_binary.c:384:34: error: this function declaration is not a prototype [-Werror,-Wstrict-prototypes]
    else if(f == 0.0f) encoded = signbit(f) ? FLOAT_NEG_ZERO : 0;
                                 ^
src/ua_types_encoding_binary.c:399:42: error: use of undeclared identifier 'INFINITY'
    else if(decoded == FLOAT_INF) *dst = INFINITY;
                                         ^
src/ua_types_encoding_binary.c:400:47: error: use of undeclared identifier 'INFINITY'
    else if(decoded == FLOAT_NEG_INF) *dst = -INFINITY;
                                              ^
src/ua_types_encoding_binary.c:402:65: error: use of undeclared identifier 'NAN'
       (decoded >= 0xff800001 && decoded <= 0xffffffff)) *dst = NAN;
                                                                ^
src/ua_types_encoding_binary.c:434:43: error: use of undeclared identifier 'INFINITY'
    else if(decoded == DOUBLE_INF) *dst = INFINITY;
                                          ^
src/ua_types_encoding_binary.c:435:48: error: use of undeclared identifier 'INFINITY'
    else if(decoded == DOUBLE_NEG_INF) *dst = -INFINITY;
                                               ^
src/ua_types_encoding_binary.c:438:83: error: use of undeclared identifier 'NAN'
       (decoded >= 0xfff0000000000001L && decoded <= 0xffffffffffffffffL)) *dst = NAN;
                                                                                  ^
8 errors generated.
```

yurivict, 2017-12-23:
Adding ```#include <math.h>``` before all includes there fixes the problem.

jpfr, 2017-12-23:
We already include math.h
https://github.com/open62541/open62541/blob/af41e659e39d4c965c5f6095c378f0a3160f9380/src/ua_types_encoding_binary.c#L315

Which version of the library are you using?

yurivict, 2017-12-23:
The revision is 36ba8b33.
I had to add these patches on FreeBSD, otherwise it didn't work on FfreeBSD.
This is not a big deal, I can just leave the patches there in the port.

```
--- src/ua_types_encoding_binary.c.orig 2017-12-20 18:00:46 UTC
+++ src/ua_types_encoding_binary.c
@@ -2,6 +2,7 @@
  * License, v. 2.0. If a copy of the MPL was not distributed with this
  * file, You can obtain one at http://mozilla.org/MPL/2.0/. */

+#include <math.h>
 #include "ua_util.h"
 #include "ua_types_encoding_binary.h"
 #include "ua_types_generated.h"
```

```
--- plugins/ua_network_tcp.c.orig       2017-12-20 18:00:46 UTC
+++ plugins/ua_network_tcp.c
@@ -29,6 +29,7 @@
 # define _WIN32_WINNT 0x0501
 #endif

+#include <unistd.h>
 #include "ua_network_tcp.h"
 #include "ua_log_stdout.h"
 #include "queue.h"
```

Pro, 2018-02-05:
In the current master source `math.h` is already included right before usage of `signbit`:
https://github.com/open62541/open62541/blob/d6c54145b49736e3e8821bff769853b1b1124b58/src/ua_types_encoding_binary.c#L311

Same for unistd.h:
https://github.com/open62541/open62541/blob/d6c54145b49736e3e8821bff769853b1b1124b58/plugins/ua_network_tcp.c#L111

I don't see any place where it would make sense to add those two additional/duplicate includes.

closed 2018-02-05 by Pro