open62541/open62541 #1482: C:\temp\godot\core\typedefs.h(33): fatal error C1083: Cannot open include file: 'stddef.h': No such file or directory

veyluna, 2017-12-24:
First of all sorry If my issue report isn't good enough.

**Operating system or device - Godot version**:

Windows 10, Godot master branch

**Issue description**:

It's the first time I'm trying to run Godot from resource code and I got this error message. In an another issue report something like this happened and resolved with a little change in the related SCsub file but even though I nearly checked all of the SCsub files I couln't find "stddef.h".

In the same issue report they also said delete the related line and after doing that another same error happened in different files.

I also tried a bunch of stuff about environment variables but I don't think it's about that.

>

Here is the full message I get:
>scons: Reading SConscript files ...
>Detected MSVC compiler: x86
>Compiled program architecture will be a 32 bit executable. (forcing bits=32).
>YASM is necessary for WebM SIMD optimizations.
>WebM SIMD optimizations are disabled. Check if your CPU architecture, CPU bits or platform are >supported!
>Checking for C header file mntent.h... no
>scons: done reading SConscript files.
>scons: Building targets ...
>[Initial build] run(["core\method_bind.gen.inc", "core\method_bind_ext.gen.inc"], 
>["core\make_binders.py"])
>[Initial build] Compiling ==> platform\windows\godot_win.cpp
>godot_win.cpp
>C:\temp\godot\core\typedefs.h(33): fatal error C1083: Cannot open include file: 'stddef.h': No such file >or directory
>scons: *** [platform\windows\godot_win.windows.tools.32.obj] Error 2
>scons: building terminated because of errors.

HolgerJeromin, 2017-12-24:
What is godot?
Did you wanted this: https://github.com/godotengine/godot

veyluna, 2017-12-24:
Ah... Yes I'm really sorry

closed 2017-12-25 by veyluna

HolgerJeromin, 2017-12-25:
Happy X-Mas :)