open62541/open62541 #1484: nodeset compiler crashes while processing minimal example xml

bobbthebuilder, 2017-12-29:
Hi,

i wanted to test the nodeset_compiler.py script that ships with open62541 from the latest available master branch and received an the following error when calling `python nodeset_compiler.py -x ~/minimal.xml ~/output.c`:
```
INFO:__main__:Preprocessing /home/user/minimal.xml
Traceback (most recent call last):
  File "nodeset_compiler.py", line 138, in <module>
    ns.addNodeSet(xmlfile, typesArray=getTypesArray(nsCount))
  File "/home/user/open62541_master/tools/nodeset_compiler/nodeset.py", line 252, in addNodeSet
    newsource = self.nodes[ref.target]
KeyError: ns=0;i=63
```

The xml i used has just a single variable node and looks as follows:

```xml
<?xml version='1.0' encoding='utf-8'?>
<UANodeSet xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">
  <NamespaceUris />
  <Aliases>
    <Alias Alias="UInt64">i=9</Alias>
    <Alias Alias="HasTypeDefinition">i=40</Alias>
    <Alias Alias="HasComponent">i=47</Alias>
  </Aliases>
  <UAVariable BrowseName="0:TestVar" DataType="UInt64" NodeId="i=20001" ParentNodeId="i=85">
    <DisplayName>TestVar</DisplayName>
    <Description>TestVar</Description>
    <References>
      <Reference IsForward="false" ReferenceType="HasComponent">i=85</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
    </References>
    <Value>
      <uax:UInt64>1000</uax:UInt64>
    </Value>
  </UAVariable>
</UANodeSet>
```
What i am doing wrong?

Stasik0, 2017-12-29:
You missed ns0 definitions. Unclude ns0 def before your xml.

bobbthebuilder, 2017-12-29:
Thanks for your quick response. It's unclear to me what "Include ns0 def before your xml." exactly means, can you show or refer to a working example?

Stasik0, 2017-12-29:
There is a whole section in the docs

https://open62541.org/doc/current/namespace_compiler.html

bobbthebuilder, 2017-12-29:
The documentation advices calling `python ./generate_open62541CCode.py <Opc.Ua.NodeSet2.xml> myNS.xml myNS`, in the current master branch the script is now named `nodeset_compiler.py` and it seems that there is no argument for <Opc.Ua.NodeSet2.xml> any more. Contrarily, in the 0.2 branch `namespaceXML` existed as parameter.

In the master branch, the README.md in `/tools/nodeset_compiler/` says calling
`python generate_open62541CCode.py /path/to/NodeSet.xml /path/to/outputfile.c` is the way to do it.

I have:
- the xml file from [https://open62541.org/doc/current/namespace_compiler.html](url)
- the namespace xml from [https://raw.githubusercontent.com/OPCFoundation/UA-Nodeset/master/Schema/Opc.Ua.NodeSet.xml](url)

How to feed `nodeset_compiler.py` so that a .c and a .h file is created?

Thank you for making that clear.

Stasik0, 2017-12-29:
Okay, the docs seem to be outdated - please open an issue in github, or better providena pull request to fix.

When in doubt - look into the cmake files that are always executed on ci platform.

https://github.com/open62541/open62541/blob/master/CMakeLists.txt#L527

Here just a one namespace (ns0) is processed.

The --xml parameter (https://github.com/open62541/open62541/blob/master/tools/nodeset_compiler/nodeset_compiler.py#L39) can take multiple xml files. Start by adding your custom xml after ns0 xml. If it works, you can remove --generate-ns0 and try to compile your file only.

You can also try to follow the tutorial with ignored nodes etc. Most likely only the command switches changed a bit.

Sorry, cannot test myself atm.

HolgerJeromin, 2017-12-29:
But we should not crash even with invalid input data :)

bobbthebuilder, 2017-12-30:
@Stasik0 i'll correct the docs and create a pull request as soon as i understand how this is currently working (and what has maybe to be fixed before).

Meanwhile i followed your hints and tried various permutations of
`$ ./nodeset_compiler.py --generate-ns0 --internal-headers --xml /path/to/Opc.Ua.NodeSet.xml ~/path/to/server_xml_from_tutorial.xml output`
getting exceptions only. Providing multiple xml files seems not to work in combination with the parameter `outputFile`.

Stasik0, 2018-01-02:
` python nodeset_compiler.py -i NodeID_NS0_Base.txt -x ../schema/Opc.Ua.NodeSet2.Minimal.xml -x test.xml test` 

then you neet to call `test()` from the generated file to create the nodes

jpfr, 2018-01-05:
Can we spell this out in the docs?

Stasik0, 2018-01-06:
Can anyone confirm?

bobbthebuilder, 2018-01-06:
The command that @Stasik0 posted worked for me and i'm about to make changes in the docs regarding this and do a pull request.

jpfr, 2018-01-07:
👍

bobbthebuilder, 2018-01-07:
Invoking ``python nodeset_compiler.py -x ../schema/Opc.Ua.NodeSet2.Minimal.xml -x test.xml test`` without the preceding ``-i NodeID_NS0_Base.txt`` and the nodeset xml from the official docs results in
```
Traceback (most recent call last):
  File "nodeset_compiler.py", line 192, in <module>
    generateOpen62541Code(ns, args.outputFile, args.suppressedAttributes, args.generate_ns0, args.internal_headers, args.typesArray, args.max_string_length)
  File "/home/useropen62541/tools/nodeset_compiler/backend_open62541.py", line 220, in generateOpen62541Code
    sorted_nodes = reorderNodesMinDependencies(nodeset)
  File "/home/user/open62541/tools/nodeset_compiler/backend_open62541.py", line 156, in reorderNodesMinDependencies
    raise Exception("Node graph is circular on the specified references. Still open nodes:\r\n" + stillOpen)
Exception: Node graph is circular on the specified references. Still open nodes:
Default Binary/ns=0;i=3062 = 1
Default XML/ns=0;i=3063 = 1
[...]
```
Is this a bug?

Pro, 2018-01-17:
To summarize: YES! We definitely need to improve the docs for the nodeset compiler (NSC)! Pull requests are welcome!

@HolgerJeromin the crash from the first post is supposed to happen. It means that somewhere in the xml a node with ID `i=63` is referenced but does not exist in the node set. I agree that there could be a better error message, but for stopping the compilation if NSC fails, I chose to throw exceptions. There are definitely better solutions, but as a quick fix, this works for now. PRs for improving the output and error handling of the NSC are greatly welcome.

@bobbthebuilder your last post is also expected behaviour. To print the nodes in the correct order (make sure that referenced nodes are there before referencing them) we need to sort the generated code. To avoid circular dependencies (which can not be solved/ordered), the server code includes a very limited basic set of nodes which are always created and necessary, independent of the used node set. The `-i NodeID_NS0_Base.txt` is exactly for this purpose: it lists the node ids of this basic set of nodes, therefore these nodes must not be generated and ignored by the NSC. If the NSC does not know that these nodes already exist it tries to order the nodes but finds a circular dependency. Therefore you get the error:

```
Exception: Node graph is circular on the specified references. Still open nodes:
Default Binary/ns=0;i=3062 = 1
Default XML/ns=0;i=3063 = 1
```

So the correct call for the minimal nodeset NS0 is (as currently created by CMake):
```
./tools/nodeset_compiler/nodeset_compiler.py --generate-ns0 --internal-headers --max-string-length=0 --ignore ./tools/nodeset_compiler/NodeID_NS0_Base.txt --xml ./tools/schema/Opc.Ua.NodeSet2.Minimal.xml ./build/src_generated/ua_namespace0
```

@bobbthebuilder I will close this issue. Anyways I'm happy to answer any further questions regarding the NSC. Just add them to the issue here and feel free to add a PR.

closed 2018-01-17 by Pro