open62541/open62541 #1485: Connection is closed by subscriptions with publishingInterval*maxKeepAliveCount > client timeout

basyskom-jvoe, 2018-01-02:
Description
===========
While adding support for subscription parameters to QtOpcUa, I have run into a problem with subscriptions in the open62541 client.

Currently, PublishRequest and PublishResponse are treated as synchronous messages (The timeout value in UA_ClientConfig_default is 5 seconds):
- A PublishRequest is sent
- The client waits up to timeout in UA_Client_Subscriptions_manuallySendPublishRequest
- If a PublishResponse is received before the timeout, everything works as expected

If a publishingInterval > timeout or a combination of publishingInterval*maxKeepAliveCount > timeout is used, there are two cases:
- The value changes before the timeout, a PublishResponse is received, everything works as expected
- There is no value change, the client runs into the timeout

For the second case, the check in ua_client.c line 324 applies and the connection to the server is closed.
According to the specification, the server can delay sending the PublishResponse until maxKeepAliveCount is reached, so this timeout should not lead to a closed connection.

I see no practicable workaround for this issue. Increasing the client timeout to a value publishingInterval*maxKeepAliveCount would block the client even longer.

IMO, the only possible solution is to use asynchronous messages for PublishRequest and PublishResponse.
Sending the publish requests could be enabled after the first successful call to UA_Client_Subscriptions_new, the next one is sent after the response for the previous request has been received.

Any ideas?

Background Information / Reproduction Steps
===========================================
Create a subscription with requestedPublishingInterval of 6000
Add a monitored item for a value that does not change often
Send a PublishRequest using manuallySendPublishRequest

Checklist
=========
Please provide the following information:

 - [X] open62541 Version (release number or git tag): 386a2f4c808c77853a8057133208554644d0047d
 - [X] Other OPC UA SDKs used (client or server): open62541 server and other servers
 - [X] Operating system: Ubuntu 16.04

StalderT, 2018-01-02:
https://github.com/open62541/open62541/issues/1332

StalderT, 2018-02-18:
Can be closed (#1497) ?

closed 2018-02-22 by basyskom-jvoe