open62541/open62541 #1488: connect to node-opcua from open62541

bnadem, 2018-01-04:
!ATTENTION!
Please read the following page carefully and provide us with all the
information requested:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Use Github Markdown to format your text:
https://help.github.com/articles/basic-writing-and-formatting-syntax/

Fill out the sections and checklist below (add text at the end of each line).

Remove this text between the ATTENTION marks after reading it.
!ATTENTION!
--------------------------------------------------------------------------------

Description
===========

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-01-05:
Hey @bnadem ,

It is possible that node-opcua uses encryption for username/password even if the SecureChannel is not encrypted.

Can you prepare a network dump with Wireshark?
It's reasonably easy to do. And we need the details to resolve the issue.

Thanks.

erossignon, 2018-01-05:
@jpfr , I confirm that node-opcua encrypts the UserIdentityToken if secure channel is not encrypted as required by the specs:
Release 1.03 26 OPC Unified Architecture, Part 4 :

> The ActivateSession Service is used to associate a user identity with a Session. When a Client provides a user identity then it shall provide proof that it is authorized to use that user identity. The exact mechanism used to provide this proof depends on the type of the UserIdentityToken. If the token is a UserNameIdentityToken then the proof is the password that is included in the token. If the token is an X509IdentityToken then the proof is a signature generated with private key associated with the Certificate. The data to sign is created by appending the last serverNonce to the serverCertificate specified in the CreateSession response. **If a token includes a secret then it should be encrypted using the public key from the serverCertificate.** 

This takes place around [here](https://github.com/node-opcua/node-opcua/blob/9e7b4a8f293a4144defb528674fe6f3c8125f798/packages/node-opcua-server/src/opcua_server.js#L983)

Let me know if you agree with this behavior or if we've missed something in interpreting the specs.

jpfr, 2018-01-05:
Hey @erossignon ,

Your side is correct. We implement encryption in PR #1282.
Decoding username/password will be implemented afterwards.

We will have to wait until then.

bnadem, 2018-01-08:
here is twork dump with Wireshark,  i hope it helpfull

bnadem, 2018-01-08:
Hey @jpfr node 
here is my network dump with Wireshark, 
[username_pcapng.txt](https://github.com/open62541/open62541/files/1611028/username_pcapng.txt)

Pro, 2018-02-01:
Closing in favour of #1539

closed 2018-02-01 by Pro

Pro, 2018-02-01:
Also duplicate of #934