open62541/open62541 #1495: Missing Tutorial for Empedded Systems with LWip and / without FreRtos

marcrupprath, 2018-01-07:
--------------------------------------------------------------------------------

Description
===========
First of all congratulation for your Project so far:
From my point of view, providing a working example of an OPC UA Server / Client based on common Microcontroller Systems like : STM32, or PIC 32MX /MZ ... would be very nice.
I have seen that you have started changing the network layer do do so.

I am currently missing working examples projects.
Proposal: 
Having a Hardwareindependet codebase is nice at fist but  fro my point of view it makes more sense having different forks / Branches .

1. For "higher" Level System, like Linux / Windows
2. One for "low" level Systems: Like FReertos or Bare Metal Based Systems: Arduino, STM32,, ---

I think the differences, at least according to the Network layer , are big enough for having different branches.

To make a long story short:
What kind of working example projects for Embedded Systems do you have ?
Can you send me a copy of them for private studies ?

jpfr, 2018-01-07:
Hello there,

the main library is platform independent. Platform-specific code lives in "plugins" that get attached to the server or client config. For example the network layer.
So I'd say we don't need branches, but rather plugins for common embedded archs.

See #1160 where our networking plugin was ported to lwip. But the changes are rather invasive.
With some work, we could fold that back into our repo as a dedicated plugin.
And cover it with continuous integration.

If you search for embedded, freertos and lwip in the github search, quite a few users in that area come up.
Anyway, contributions for embedded support, also bug reports and general experiences, are very welcome.

marcrupprath, 2018-01-13:
Hello,
if you are looking for a good Embedded OPC UA Reference in Hardware and Software i suggest the following:

https://www.infineon.com/cms/de/product/evaluation-boards/kit_xmc48_relax_ecat_v1/
with
OPC UA Example impelemntation; 
https://www.infineon.com/cms/en/product/promopages/aim-mc/dave_downloads.html
---> OPCUASERVER_LWIP_XMC47

idea is to replace given Demo lib with open 62541

Hope that helps ;-)

SHveg, 2018-04-12:
@marcrupprath 

Hi Marc, have you tried using the open62541 on Infineon XMC4700?
I would appreciate if you could share your experience regarding that.

Best Regards!

Pro, 2018-04-12:
Please note that there is now an architecture plugin structure: #1595 
It will be merged, after all the PubSub batches are merged, so hopefully quite soon.

Based on that it should be easier to port open62541 to specific architectures

marcrupprath, 2018-04-13:
@SHVeg;
Hello,
unfortunately not.
I am curently very bussy with other  projects
OPC UA, expecially on embedded systems,  acctually  is just a private hobby ;-)
One of many, and to less time for all of them ;-)

I don't think that i will have time to continue my investigations during this year.
I will come back when i have new news.

Greatings  Marc

Pro, 2018-11-06:
Examples for other architectures like FreeRTOS or ESP32 are now in the master branch (added with #1595 and follow ups).

closed 2018-11-06 by Pro