open62541/open62541 #1498: OPCUA STATUS XOR VALUE

jldmatwork, 2018-01-08:
I don't understand this code:
If you change status to different from good, value has no sense, why? 
¿what if i want to recover old status before disconnection for example.????
How can i do that in other way.

/* Used in inline functions exposing the Read service with more syntactic sugar
 * for individual attributes */
UA_StatusCode
__UA_Server_read(UA_Server *server, const UA_NodeId *nodeId,
                 const UA_AttributeId attributeId, void *v) {
    /* Call the read service */
    UA_ReadValueId item;
    UA_ReadValueId_init(&item);
    item.nodeId = *nodeId;
    item.attributeId = attributeId;
    UA_DataValue dv = UA_Server_read(server, &item, UA_TIMESTAMPSTORETURN_NEITHER);

    /* Check the return value */
    UA_StatusCode retval = UA_STATUSCODE_GOOD;
    if(dv.hasStatus)
        retval = dv.hasStatus;
    else if(!dv.hasValue)
        retval = UA_STATUSCODE_BADUNEXPECTEDERROR;
    if(retval != UA_STATUSCODE_GOOD) {
        UA_DataValue_deleteMembers(&dv);
        return retval;
    }
......
.....
}

jpfr, 2018-01-08:
The read request can contain many read operations.
So there is an overall status code for the service and one for each individual operation result.

Since this "convenience method" reads one attribute only, we combine the status of the service and the single read operation.

jldmatwork, 2018-01-08:
I think i didn't explain myself ok. Or i don't understand you.

I agree that it could be a general status and an attribute status, but the problem is that if the service is OK, and you set a (f.e. boolean) variable status to BAD_DISCONNECT with value false.

Then if you read that variable, you get the UA_StatusCode BAD_DISCONNECT , but you don't have the value because of those lines i copy before.
      `retval = dv.hasStatus;----------->  retval = 1;`      
As retval is different from GOOD (0) it returns.
`if(retval != UA_STATUSCODE_GOOD) {
        UA_DataValue_deleteMembers(&dv);
        return retval;
    }`

jpfr, 2018-01-09:
Thanks for precising. This has already been fixed on master.
https://github.com/open62541/open62541/blob/master/src/server/ua_services_attribute.c#L465

You should be able to upgrade to the master branch or a nightly build (https://open62541.org/releases/) with no troubles. Please keep up reporting regressions!

closed 2018-01-09 by jpfr