open62541/open62541 #1499: adding multiple monitored items

mukunth1987, 2018-01-08:
!ATTENTION!

!ATTENTION!
--------------------------------------------------------------------------------

Description
===========
I was trying to add multiple monitored items within single subscription context.

But UA_Client_Subscriptions_addMonitoredItem() allows to add only one monitorItem to be added. Which means, I need to call UA_Client_Subscriptions_addMonitoredItem() API multiple times for adding more monitorItems. 

Digging into the code, I found "itemsToCreate" and "itemsToCreateSize" variables in UA_CreateMonitoredItemsRequest structure. But itemsToCreateSize is always set to 1 inside UA_Client_Subscriptions_addMonitoredItem() API.

Is there a way to add multiple monitoredItems in a single request call ?? 

FYI,  I am using single header/source files for building.

Background Information / Reproduction Steps
===========================================
Checkout master and run CTT for client.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

WoutvdBerg, 2018-01-08:
It's implemented since #1490.

jpfr, 2018-01-08:
A single-file distribution from the current master branch can be downloaded here:
https://open62541.org/releases/

closed 2018-01-08 by jpfr