open62541/open62541 #1501: Extract UA_Argument from UA_ExtensionObject

gerdBert, 2018-01-12:
Description
===========

I want to to read the input arguments of a method with a client that has been built with _open62541_. Therefore, I browse the nodes connected to the method and receive two Properties called _InputArguments_ and _OutputArguments_.
The Value of the Proptery _InputArguments_ is of type _UA_ExtensionObject_ (when looking into the  debugger) and should contain an array of the type _UA_Argument_. 

I would like to extract the elements of type _UA_Argument_ holded by the _UA_ExtensionObject_. Converting the value of the Property to an _UA_ExtensionObject_ works fine. The _typeName_ of that Object is "Argument" as expected. Converting the data of the _UA_ExtensionObject_ to _UA_Argument_ leads to garbage data.

Somehow it must be possible:  in UAexpert I can see the whole content of the value of the Property _InputArguments_(refer to attached screenshot).
[image]

Background Information / Reproduction Steps
===========================================
```cpp
UA_Variant value;
UA_Variant_init(&value);
UA_Client_readValueAttribute(client, InputArgumentsNodeId, value);

UA_ExtensionObject* pInputArguments= reinterpret_cast<UA_ExtensionObject*>(value.data);
UA_Argument* pArgument1 = reinterpret_cast<UA_Argument*>(pInputArguments->content.decoded.data);
```
pInputArguments seems to be correctly reinterpreted. E.g. The _typeName_ is "Argument" . But when I look into pArgument1 the values make no sense.

The following statement is true
```cpp
value.type->typeIndex == UA_TYPES_EXTENSIONOBJECT
UAstrcmp(pInputArguments.type->typeName,"Argument") == 0
````

Checklist
=========
 - [x] open62541 Version (release number or git tag): v0.2
 - [x] Other OPC UA SDKs used (client or server): UAexpert for verification
 - [x] Operating system: Win10

jpfr, 2018-01-14:
Hmm. This looks correct. I don't see an obvious error.
Sorry if we can't be of better help.

Sometimes DrMemory helps in finding memory-related bugs.
http://drmemory.org/

Infinity95, 2018-01-14:
Is it possible to supply a hexdump of the value data after you read it?

gerdBert, 2018-01-16:
I must have been overseen something. I retested it and now the conversion is working as expected.
→ issue can be closed

ptr to UA_ExtensionObject
[image]

ptr to Argument X1 one of type UA_Argument
[image]

Thank you for your advice!

closed 2018-01-16 by gerdBert