open62541/open62541 #1502: issue faced while trying to compile "myServer.c"

issararab, 2018-01-12:
While running the following command => "gcc -std=c99 open62541.c myServer.c -o myServer"

I got the bellow warnings and errors:

myServer.c: In function 'main':
myServer.c:36:5: warning: implicit declaration of function 'UA_ServerConfig_new_default' [-Wimplicit-function-declaration]
     UA_ServerConfig *config = UA_ServerConfig_new_default();
     ^
myServer.c:36:31: warning: initialization makes pointer from integer without a cast
     UA_ServerConfig *config = UA_ServerConfig_new_default();
                               ^
myServer.c:37:25: error: incompatible type for argument 1 of 'UA_Server_new'
     UA_Server *server = UA_Server_new(config);
                         ^
In file included from myServer.c:24:0:
open62541.h:9816:23: note: expected 'UA_ServerConfig' but argument is of type 'struct UA_ServerConfig *'
 UA_Server UA_EXPORT * UA_Server_new(const UA_ServerConfig config);
                       ^
myServer.c:41:5: warning: implicit declaration of function 'UA_ServerConfig_delete' [-Wimplicit-function-declaration]
     UA_ServerConfig_delete(config);
     ^

Can someone help with this issues?
Thanks

HolgerJeromin, 2018-01-12:
What is the content of myServer.c?
Have you seen 
https://github.com/open62541/open62541/blob/master/examples/server.c

issararab, 2018-01-14:
The content of the file is the same as the one of "tutorial_server_firststeps.c" 
I followed the instructions written in the file to launch the server.

Here is it:

#include <signal.h>
#include "open62541.h"

UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server *server = UA_Server_new(config);

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}

Stasik0, 2018-01-14:
Did you mix the master and 0.2 braches? Try to pick the open62541.* files from https://open62541.org/releases/. For example, https://open62541.org/releases/a5d9a1f7bc.zip .

Pro, 2018-01-17:
This definitely looks like you are mixing 0.2 and master.

`UA_ServerConfig *config = UA_ServerConfig_new_default();` is from the master branch and the error message tells you that the method `UA_ServerConfig_new_default();` does not exist.

In 0.2 we had `UA_ServerConfig config = UA_ServerConfig_standard;`

To fix your issue use one of the compiled master versions mentioned in previous posts.

closed 2018-01-17 by Pro