open62541/open62541 #1511: Missing User forum

marcrupprath, 2018-01-16:
Hello,
i am missing a common user forum where we can share our experiences and for asking questions about the usage with yout library:

I think this is not the right place for question / answers, right ?
So how to address someone with "common" , "daily use" questions ?

I start with first question:
How to customize the code according to the sources shown in the plugin example ?
Because.
I havent found the position how the refrence from open62541.c /open62541.h is made to the sources like
ua_network_tcp.h and ua_network_tcp.c for example.

Thanks for your helping hand ?

jpfr, 2018-01-16:
Hello Marc,

you can ask these questions here or on the mailing list.
We try to run a *lightweight* organisation with little overhead.
Is there another open source project from whom we can learn in this regard?

If you want to customize plugins, I recommend not to use the single-file distribution but build from source. The build is encoded in the CMakeLists.txt files. It's copy/paste in moste cases.

Pro, 2018-01-17:
Here is the link to the mailing list: https://groups.google.com/forum/#!forum/open62541

Feel free to ask your questions there or here on github. If you already opened a topic and have a completely unrelated question it's best if you create a new thread/issue to keep the discussions separated.

We are happy for any feedback and discussions!

closed 2018-01-17 by Pro