open62541/open62541 #1513: adding indexRange to monitoredItems

WoutvdBerg, 2018-01-17:
I would like to implement indexRanges to monitoredItems.
Since there are many changes to the monitored items ongoing, is it possible at this stage?

Are there any obstructions?

jpfr, 2018-01-17:
This is already implemented.
If indexranges don't work, file a bug.

WoutvdBerg, 2018-01-17:
Can't find them in client side use with UA_Client_subscriptions_addMonitoredItem.

jpfr, 2018-01-17:
UA_Client_Subscriptions_addMonitoredItems lets you define a `UA_MonitoredItemCreateRequest`.
This contains an `itemToMonitor` that carries the indexRange.

If you have an idea for a nicer API, proposals are welcome.

WoutvdBerg, 2018-01-17:
Thanks, I was still looking in the 'old' addMonitoredItem function. Issue resolved

closed 2018-01-17 by WoutvdBerg