open62541/open62541 #1520: Setting Monitoring Mode in

ChaJiWon, 2018-01-18:
Description
===========
Hi! I read about *Monitoring Mode* in service specification.

> Disabled
*The item being monitored is not sampled or evaluated, and Notifications are not generated or queued. Notification reporting is disabled.*

> REPORTING
*The item being monitored is sampled and evaluated, and Notifications are generated and queued. Notification reporting is enabled.*

So, I think it needs to add some more code to open62541.

Background Information / Reproduction Steps
===========================================
I want to suggest below codes.

*ua_services_subscription.c*

```
In Operation_SetMonitoringMode

.....

if(mon->monitoringMode == UA_MONITORINGMODE_REPORTING) {
        MonitoredItem_registerSampleCallback(server, mon);
    } else if (mon->monitoringMode == UA_MONITORINGMODE_DISABLED){
        /*  Setting the mode to DISABLED causes all queued Notifications to be delete */
    	MonitoredItem_queuedValue *val, *val_tmp;
		TAILQ_FOREACH_SAFE(val, &mon->queue, listEntry, val_tmp) {
			TAILQ_REMOVE(&mon->queue, val, listEntry);
			UA_DataValue_deleteMembers(&val->value);
			UA_free(val);
		}
		mon->currentQueueSize = 0;
        /* initialize lastSampledValue */
		mon->lastSampledValue.data = NULL;
		mon->lastSampledValue.length = 0;

		MonitoredItem_unregisterSampleCallback(server, mon);
    }

```

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): v0.2 release
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 14.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-01-18:
👍 
If you send us a pull request, we can merge it right away.
You can also reference the relevant spec section in a comment for future reference.

ChaJiWon, 2018-01-24:
#1522

Pro, 2018-02-01:
Closing in favor of #1522

PR #1522 (2018-02-23): Additional logic of setting monitoring mode
fix #1520  issue.

This commit is related with service specification (Part4)

```
7.18 Monitoring Mode

DISABLED_0
The item being monitored is not sampled or evaluated, and Notifications are not generated or queued. Notification reporting is disabled.

REPORTING_2
The item being monitored is sampled and evaluated, and Notifications are generated and queued. Notification reporting is enabled.
```
closed 2018-02-01 by Pro