open62541/open62541 #1527: Maximum chunk count not checked

Infinity95, 2018-01-19:
Description
===========
Currently (as far as I know) the server and client don't check if the maximum chunk count was reached while receiving. This could lead to denial of service attacks.
Also I'm not quite sure what the standard says about chunks that are never completed.
Let's say for example that a client keeps sending intermediate chunks with different request ids. We will at one point reach the maximum chunk count and then our chunks list is full, but is then never cleared so no more messages can be received from that point onwards. Does the standard say anything about some kind of timeout for uncompleted chunks? I haven't found anything yet.
This could be especially bad if no maximum chunk count is configured, since we could keep sending intermediate chunks with different request ids and fill up the server memory.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): master

ChHaa, 2018-03-06:
Standard has no information about that afaik, was still an open thing to test within the BSI study on OPC UA security. So feel free to think about things that could go wrong ;)

Infinity95, 2018-03-19:
So i checked and it's possible to send any amount of chunks to a specific SecureChannel without sending the final chunk.
Even if a maximum Chunk count and a maximum message size is configured it's still possible to send 2³² intermediate chunks with different request ids.
At 8kb maximum message size this amounts to around 30 TB of data (if i calculated correctly).
The chunks in open62541 currently are only cleared up if the channel is closed.
So if the channel is kept open the server can be flooded with data until it is out of memory.

ChHaa, 2018-03-20:
That would mean we can only send and process different messages sequentially, is that ok with everyone ? Would open another DoS Vector as we can block any communication with the server by defining and sending really large messsages, right ?

Infinity95, 2018-03-20:
Wouldn't that be mitigated by the maximum message size?
The messages only have to be sequential for one SecureChannel. Two separate SecureChannels can be completely independent of each other I believe.

ChHaa, 2018-03-20:
I was talking about if you have two independent messages that have to be send to a server by the same securechannel endpoint. We can only send them sequentially then, that might be ok and then your fix-idea would mitigate a lot of problems.

jpfr, 2020-07-07:
Fixed since quite some time ago.

closed 2020-07-07 by jpfr