open62541/open62541 #1533: Browsing with client issue

fixthefx, 2018-01-22:
Description
===========
When browsing example server nodes with the client I see reference types like "FolderType" (example output below). Is it possible to get only node names?

```
Browsing nodes in objects folder:
NAMESPACE NODEID           BROWSE NAME      DISPLAY NAME
0         61               FolderType       FolderType
0         2253             Server           Server
1         the.answer       the answer       the answer
```
If browse with UAExpert "FolderType" is not presented in the tree.

The second problem is when try to browse Server nodes recursively. Then client prints 20-30 rows and exit with exception.

```
Browsing nodes in objects folder:
NAMESPACE NODEID           BROWSE NAME      DISPLAY NAME
0         61               FolderType       FolderType
0         11564            OperationLimitsType OperationLimitsType
0         11574            MaxMonitoredItemsPerCall MaxMonitoredItemsPerCall
0         68               PropertyType     PropertyType
0         80               Optional         Optional
0         77               ModellingRuleType ModellingRuleType
0         111              NamingRule       NamingRule
0         68               PropertyType     PropertyType
0         78               Mandatory        Mandatory
0         77               ModellingRuleType ModellingRuleType
0         111              NamingRule       NamingRule
0         68               PropertyType     PropertyType
0         78               Mandatory        Mandatory
0         77               ModellingRuleType ModellingRuleType
0         111              NamingRule       NamingRule
0         68               PropertyType     PropertyType
0         78               Mandatory        Mandatory
0         77               ModellingRuleType ModellingRuleType
0         111              NamingRule       NamingRule
0         68               PropertyType     PropertyType
0         78               Mandatory        Mandatory
0         77               ModellingRuleType ModellingRuleType
0         111              NamingRule       NamingRule
0         68               PropertyType     PropertyType
0         78               Mandatory        Mandatory
0         77               ModellingRuleType ModellingRuleType
0         111              NamingRule       NamingRule
0         68               PropertyType     PropertyType
0         78               Mandatory        Mandatory
0         77               ModellingRuleType ModellingRuleType
0         111              NamingRule       NamingRule
0         68               PropertyType     PropertyType
0         78               Mandatory        Mandatory
0         77               ModellingRuleType ModellingRuleType
0         111              NamingRule       NamingRule
0         68               PropertyType     PropertyType
0         78               Mandatory        Mandatory
0         77               ModellingRuleType ModellingRuleType
0         111              NamingRule       NamingRule
0         68               PropertyType     PropertyType
0         78               Mandatory        Mandatory
0         77               ModellingRuleType ModellingRuleType
0         111              NamingRule       NamingRule
0         68               PropertyType     PropertyType
0         78               Mandatory        Mandatory
0         77               ModellingRuleType ModellingRuleType
0         111              NamingRule       NamingRule
```

Recursive client example

```
void listTreeRecursive(UA_Client *client, UA_NodeId nodeId)
{
  size_t i, j;

  /* Browse some objects */
  UA_BrowseRequest bReq;
  UA_BrowseRequest_init(&bReq);
  bReq.requestedMaxReferencesPerNode = 0;
  bReq.nodesToBrowse = UA_BrowseDescription_new();
  bReq.nodesToBrowseSize = 1;

  UA_NodeId_copy(&nodeId, &bReq.nodesToBrowse[0].nodeId);
  bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL; /* return everything */

  UA_BrowseResponse bResp = UA_Client_Service_browse(client, bReq);

  for ( i = 0; i < bResp.resultsSize; i++)
  {
      for ( j = 0; j < bResp.results[i].referencesSize; j++)
      {
        UA_ReferenceDescription *ref = &(bResp.results[i].references[j]);

          if(ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_NUMERIC)
          {
              printf("%-9d %-16d %-16.*s %-16.*s\n", ref->nodeId.nodeId.namespaceIndex,
                     ref->nodeId.nodeId.identifier.numeric, (int)ref->browseName.name.length,
                     ref->browseName.name.data, (int)ref->displayName.text.length,
                     ref->displayName.text.data);

              listTreeRecursive(client, UA_NODEID_NUMERIC(ref->nodeId.nodeId.namespaceIndex,
                      ref->nodeId.nodeId.identifier.numeric));

          }
          else if(ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_STRING)
          {
              printf("%-9d %-16.*s %-16.*s %-16.*s\n", ref->nodeId.nodeId.namespaceIndex,
                     (int)ref->nodeId.nodeId.identifier.string.length,
                     ref->nodeId.nodeId.identifier.string.data,
                     (int)ref->browseName.name.length, ref->browseName.name.data,
                     (int)ref->displayName.text.length, ref->displayName.text.data);

              listTreeRecursive(client, UA_NODEID_STRING(ref->nodeId.nodeId.namespaceIndex,
                      (char*)ref->nodeId.nodeId.identifier.string.data));
          }
          /* TODO: distinguish further types */
      }
  }

  UA_BrowseRequest_deleteMembers(&bReq);
  UA_BrowseResponse_deleteMembers(&bResp);
}

int main(int argc, char **argv)
{
    UA_Client *client = UA_Client_new(UA_ClientConfig_default);
    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://192.168.0.17:4840");
    //UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://192.168.0.52:4841");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return (int)retval;
    }

    printf("Browsing nodes in objects folder:\n");
    printf("%-9s %-16s %-16s %-16s\n", "NAMESPACE", "NODEID", "BROWSE NAME", "DISPLAY NAME");
    listTreeRecursive(client, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER));

    printf("Program End\n");

    /* Clean up */
    UA_Client_delete(client); /* Disconnects the client internally */
    return UA_STATUSCODE_GOOD;
}

```

The same test with python OPC UA client prints correct nodes and there is not exception.

How can I skip showing this reference types and what could cause this exception?

Thank you!

Background Information / Reproduction Steps
===========================================
The idea is to make aggregated server which represent another server behind.

1. Compile and start test server:
https://github.com/open62541/open62541/blob/master/examples/tutorial_server_variable.c
2. Compile and run example client code from this issue.

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): Master
 - [x] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Tested on Windows and Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-01-23:
Hello there,

1. You can select which attributes to return in the BrowseDescription. See the documentation and/or Part 4 of the spec.
2. You can select the referencetypes to follow. Probably UAExpert has different settings so that the FolderType does not show up. The reference with reference-type `hasTypeDefinition` is however valid.
3. Crashs shouldn't happen. Can you run with valgrind or Dr. Memory (if on Windows) for detailed reports? Otherwise, a debugging stracktrace will also help us.

StalderT, 2018-01-23:
>  and what could cause this exception?

Due to a deep recursive loop (stack overflow) : 

```
==994== Stack overflow in thread #1: can't grow stack to 0x1ffe801000
==994== 
==994== Process terminating with default action of signal 11 (SIGSEGV)
==994==  Access not within mapped region at address 0x1FFE801FE8
==994== Stack overflow in thread #1: can't grow stack to 0x1ffe801000
==994==    at 0x118181: NodeId_decodeBinary (in /data/open62541/a.out)
==994==  If you believe this happened as a result of a stack
==994==  overflow in your program's main thread (unlikely but
==994==  possible), you can try to increase the size of the
==994==  main thread stack using the --main-stacksize= flag.
==994==  The main thread stack size used in this run was 8388608.
==994== Stack overflow in thread #1: can't grow stack to 0x1ffe801000
==994== 
==994== Process terminating with default action of signal 11 (SIGSEGV)
==994==  Access not within mapped region at address 0x1FFE801FC8
==994== Stack overflow in thread #1: can't grow stack to 0x1ffe801000
==994==    at 0x4A266B0: _vgnU_freeres (vg_preloaded.c:59)
==994==  If you believe this happened as a result of a stack
==994==  overflow in your program's main thread (unlikely but
==994==  possible), you can try to increase the size of the
==994==  main thread stack using the --main-stacksize= flag.
==994==  The main thread stack size used in this run was 8388608.
```

fixthefx, 2018-01-26:
Thank you for the help!

closed 2018-02-01 by Pro