open62541/open62541 #1535: Finding Datatype from Server

crazyblack4, 2018-01-23:
Hi,
first of all sorry for kid question. I want to display datatype of each node present in Namespace in client side. for that i used the below code
```
UA_NodeId id;
id.namespaceIndex = ref->nodeId.nodeId.namespaceIndex;
id.identifierType=UA_NODEIDTYPE_STRING;
id.identifier.string.data=ref->nodeId.nodeId.identifier.string.data;
/*Datatype Read*/
        const UA_DataType * UA_findDataType(const UA_NodeId *id);
```
but this Datatype Read API always returning **Null** for me. Please help me to resolve.

i'm using **open62541 v0.2**

Pro, 2018-01-23:
The `UA_findDataType` method can be used to get a datatype description of a built in OPC UA type.
e.g. the OPC UA Specification defines the type ID for `QualifiedName` as `namespace=0, id=20`.
To get additional information about that datatype you can use the `UA_findDataType` method:

```
    UA_NodeId id=UA_NODEID_NUMERIC(0,20);
    const UA_DataType *dataType;
    dataType=UA_findDataType(&id);
```

Since you prvided some arbitrary string id, there is no known datatype represented by this ID and you get NULL.

What exactly do you want to achieve?

jpfr, 2018-01-23:
Hello @crazyblack4 ,

probably you want to use the "Browse" service.
https://open62541.org/doc/0.2/server.html#browsing

However, browse is hierarchical starting at the Root node.
You have to use it several times to get the complete graph.

Please reopen if required.

closed 2018-01-23 by jpfr

crazyblack4, 2018-01-24:
Hi @Pro @jpfr 

i'm sharing here a screenshot of UAExpert 
[image]

from the above image i'm able to get the datatype of NodeId(red box). But i need to get the datatype of variable which is in green box. So i can read/write easily wrt datatype.

In the server side i have set the variable like below  
```
    UA_VariableAttributes FloatVar;
    UA_VariableAttributes_init(&FloatVar);
    UA_Float myFloat = 3.14;
    UA_Variant_setScalarCopy(&FloatVar.value, &myFloat, &UA_TYPES[UA_TYPES_FLOAT]);
    FloatVar.dataType = UA_TYPES[UA_TYPES_FLOAT].typeId;    ////<<-----This is the one i needed to find from client 
    FloatVar.description = UA_LOCALIZEDTEXT_ALLOC("en_US","the float");
    FloatVar.displayName = UA_LOCALIZEDTEXT_ALLOC("en_US","the float");
    FloatVar.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    UA_NodeId myFloatNodeId = UA_NODEID_STRING(1, "the.float");
    UA_NodeId parentmyFLoatNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferencemyFloatNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId myFloatVariableType = UA_NODEID_NULL;
    UA_QualifiedName myfloatName = UA_QUALIFIEDNAME_ALLOC(1, "the float");
    
    UA_Server_addVariableNode(opc_server, myFloatNodeId, parentmyFLoatNodeId,
                              parentReferencemyFloatNodeId, myfloatName,
                              myFloatVariableType, FloatVar, NULL, NULL);
    
    UA_Variant_deleteMembers(&FloatVar.value);
```

Please correct if i'm going wrong direction.

Thank you.

basyskom-jvoe, 2018-01-24:
The information you are looking for is in the DataType attribute of Variable nodes.
To read this attribute, you have to use UA_Client_readDataTypeAttribute(). This returns the node id of the data type definition of the node's Value attribute.

You can put this node id into UA_findDataType() to get the name and other information about the type.

crazyblack4, 2018-01-24:
Hello @basyskom-jvoe  @Pro  @jpfr  

Great. Thank you guys.

```
 void ReadDatatype(UA_Client *client, UA_NodeId nodeId){

    UA_StatusCode status;
    UA_NodeId dataType;
    UA_NodeId_init(&dataType);

    status = UA_Client_readDataTypeAttribute(client, nodeId, &dataType);

    if(UA_STATUSCODE_GOOD != status) {
	  printf("Err: UA_Client_readDataTypeAttribute()->status=%x =>", status);
    }
    cout<<"DataType Attribute of the given NodeId is :"<<dataType.identifier.numeric<<endl;
}
```
call from main()
```
    const UA_NodeId nodeId=UA_NODEID_NUMERIC(1,44);
    ReadDatatype(client, nodeId);
    
    const UA_NodeId nodeId1=UA_NODEID_STRING(1,"the.bool");
    ReadDatatype(client, nodeId1);
```