open62541/open62541 #1539: Basic128Rsa15

mmesa2016, 2018-01-23:
How write UA client Basic128Rsa15 example. To connect server (ej: Kepware Server)

jpfr, 2018-01-23:
Encryption support landed only a few days ago for the server.
Client support is incomplete.

This requires two enhancements:
- Add a list of supported SecurityPolicies to the client configuration
- Select the endpoint with the desired securitypolicy when opening the SecureChannel

The server code should be a good reference for using the SecureChannel layer with encryption.

srdgame, 2018-01-24:
@jpfr Is there a planned date for supporting Basic128Rsa15 in client?

forkmaster7, 2018-02-07:
@jpfr Can you please give me a short description on it. 
After selecting the endpoint with desired security policy, is the client going to use the same implementation of `UA_SecurityPolicy_Basic128Rsa15` function or the functionality needs to be changed.
Or any specific suggestion are welcome!

Pro, 2018-02-07:
@Infinity95 can you quickly summarize/sketch what would need to be done, maybe we find a volunteer (@forkmaster7?) to implement this?

jpfr, 2018-02-08:
@forkmaster7 

The client configuration would have to contain a list of securitypolicies.
These are the same policies already implemented for the server.

The actual policy for a SecureChannel is chosen when the endpoint is selected.

Infinity95, 2018-02-08:
The client would need some changes to how it handles securechannels.
It is also missing functionality for selecting securitypolicies. The discovery service for example always needs securitypolicy none and it should be possible for the user to just connect to a specified endpoint without much work. This means the required policy should be selected automatically.
It's just a matter of making a clean interface for that and integrating the already implemented functionality like certificate verification.
A lot of work is already done by the securechannel but theres some verification stuff that needs to be implemented by the client (see the standard).

Pro, 2018-03-26:
Partially implemented in #1686

jpfr, 2019-02-15:
Implemented on master.

closed 2019-02-15 by jpfr