open62541/open62541 #1541: Limits of Publishing Interval & Sampling Interval

ChaJiWon, 2018-01-24:
Description
===========

I saw the code limits publisng interval / sampling interval value in ua_config_defaults.c.
I think it means open62541 ensures monitoring times that 100ms becuase minimuj publisng interval value is 100ms.

```
    /* Limits for Subscriptions */
    conf->publishingIntervalLimits = UA_DURATIONRANGE(100.0, 3600.0 * 1000.0);

   ......

    /* Limits for MonitoredItems */
    conf->samplingIntervalLimits = UA_DURATIONRANGE(50.0, 24.0 * 3600.0 * 1000.0);
```

Does it have any special reason or standard to set publishing interval values that 100ms?

Background Information / Reproduction Steps
===========================================

I set minimum sampling interval to 10ms of server nodes. And modify value of server nodes every 10ms.

Next, I changed minimum value of sampling interval / publishing interval to 10ms and requested subscription / monitoring item service with 10ms of sampling interval . And I saw monitoring well every 10ms.

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-01-24:
> Does it have any special reason or standard to set publishing interval values that 100ms?

We can probably set both numbers to 10ms. Anybody not okay with that?
The only downside I see is the possibility for malicious clients generating a lot of load.

In addition, we should add more documentation for the server settings.
The settings in general and the defaults + their rationale.

StalderT, 2018-01-24:
> We can probably set both numbers to 10ms. Anybody not okay with that?

If I use this server : [server.txt](https://github.com/open62541/open62541/files/1659056/server.txt)

Monitoring 10'000 items with UaExpert (sampling interval = 100ms and publishing interval = 100ms) my server use 20% of the CPU.

Monitoring 10'000 items with UaExpert (sampling interval = 10ms and publishing interval = 10ms) my server use 100% of the CPU.

Personally I think it's not a good idea to change that. On the other hand we can make public ```UA_DURATIONRANGE```. If someone really wants to go faster he can do it by changing the configuration before calling ```UA_Server_new```.

```
    config->publishingIntervalLimits = UA_DURATIONRANGE(10.0, 3600.0 * 1000.0);
    config->samplingIntervalLimits = UA_DURATIONRANGE(10.0, 24.0 * 3600.0 * 1000.0);
    UA_Server *server = UA_Server_new(config);
```

jpfr, 2018-01-24:
The server configuration will now be added to the html/pdf documentation.
#1542

The config is already "public API". We could add a sentence or two, so that users understand the downsides of setting the limits very low.

ChaJiWon, 2018-01-24:
Okay. I understand. Thank you:-))

closed 2018-01-24 by ChaJiWon