open62541/open62541 #1547: Duplicated Nodes after Modeling

lehned, 2018-01-24:
Description
===========
In a test, i made with _UA Modeller_ a object type _batype_ , containing variable _at_ with TypeDefinition _AnalogItemType_ (_EURange_ property). This object type is used by an object instance _myb_ as _type definition_. This test model is exported as XML File, describing the new nodes for the _address space_. The nodeset compiler converts this to files _testmodel.c/.h_ which is included in test program _test.c_.

Problem:
In _UA Expert_ browsing the object _myb_ contains duplicated entities for each component , once the one described by XML (6009) and the same with unknown node (ID's 53xxx). Also in object type browsing all components are duplicated, once from XML and once from unknown node (ID 53xxx).

It seems the open62541 stack generates additional nodes according to the _type definition_, which is in conflict to the xml model which explicitly describes this additional nodes for an object related to the their _type definition_.

What is the right way to handle this? 

Thanks for your great work !

Background Information / Reproduction Steps
===========================================

Open62541 Version 0.3 dev - last log : 386a2f4c808c77853a8057133208554644d0047d

1. compile model with `python ../tools/nodeset_compiler/nodeset_compiler.py -e ../deps/Schema/Opc.Ua.NodeSet2.xml -x ../module/issue_dup_entries.xml ../module/testmodel` required FULL_NS0
```
<?xml version="1.0" encoding="utf-8"?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/issue_dup_entries/Types.xsd" xmlns:ua="http://unifiedautomation.com/Configuration/NodeSet.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/issue_dup_entries/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="UInt16">i=5</Alias>
        <Alias Alias="Organizes">i=35</Alias>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasProperty">i=46</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
        <Alias Alias="Range">i=884</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ua:ModelInfo Tool="UaModeler" Hash="SNjhHbijDY5/AQOIat8h9A==" Version="1.5.2"/>
        </Extension>
    </Extensions>
    <UAObjectType NodeId="ns=1;i=1003" BrowseName="1:batype">
        <DisplayName>batype</DisplayName>
        <References>
            <Reference ReferenceType="HasComponent">ns=1;i=6003</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
        </References>
    </UAObjectType>
    <UAVariable DataType="UInt16" ParentNodeId="ns=1;i=1003" NodeId="ns=1;i=6003" BrowseName="1:at" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>at</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=2368</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=6004</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1003</Reference>
        </References>
    </UAVariable>
    <UAVariable DataType="Range" ParentNodeId="ns=1;i=6003" NodeId="ns=1;i=6004" BrowseName="EURange">
        <DisplayName>EURange</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=6003</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
    </UAVariable>
    <UAObject NodeId="ns=1;i=5003" BrowseName="1:myb">
        <DisplayName>myb</DisplayName>
        <References>
            <Reference ReferenceType="HasComponent">ns=1;i=6009</Reference>
            <Reference ReferenceType="HasTypeDefinition">ns=1;i=1003</Reference>
            <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
        </References>
    </UAObject>
    <UAVariable DataType="UInt16" ParentNodeId="ns=1;i=5003" NodeId="ns=1;i=6009" BrowseName="1:at" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>at</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=2368</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=5003</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=6010</Reference>
        </References>
    </UAVariable>
    <UAVariable DataType="Range" ParentNodeId="ns=1;i=6009" NodeId="ns=1;i=6010" BrowseName="EURange">
        <DisplayName>EURange</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=6009</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
    </UAVariable>
</UANodeSet>
```
2. create cmake config `vi open62541-config.cmake`
```
set(UA_ENABLE_AMALGAMATION ON CACHE BOOL "")
set(UA_ENABLE_METHODCALLS ON CACHE BOOL "")
set(UA_ENABLE_GENERATE_NAMESPACE0 ON CACHE BOOL "")
set(CMAKE_BUILD_TYPE Debug CACHE STRING "")
set(UA_ENABLE_FULL_NS0 ON CACHE BOOL "")
```
2. compile _open62541_, to generate amalgamation files _open62541.c/.h_
```
cd open62541
rm -rf CMakeFiles
rm -f CMakeCache.txt
cmake -C ../open62541-config.cmake .
```
3. compile testprogram _test.c_
```
#include <stdint.h>
#include "open62541.h"
#include "testmodel.h"

struct st_private_opcua_data
{
  UA_ServerConfig       *config;
  UA_ServerNetworkLayer nl;
  UA_Server             *server;
  volatile UA_Boolean   running;
  uint32_t              ulPort;
};

struct st_private_opcua_data g_stOPCUA;

int main(int argc, char **argv) {

	struct st_private_opcua_data *pPriv = &g_stOPCUA;

    /* Create a server with one network layer listening on port 4840 */
    pPriv->config = UA_ServerConfig_new_minimal((pPriv->ulPort==0)?4840:pPriv->ulPort, NULL);
    //UA_ServerConfig_set_customHostname(pPriv->config, UA_STRING_ALLOC(cURI));
    /** load server, initiate new server instance */
    pPriv->server = UA_Server_new(pPriv->config);

    testmodel(pPriv->server);

    pPriv->running = 1;

    /* Run the server loop */
    switch (UA_Server_run(pPriv->server, &pPriv->running))
    {
      case UA_STATUSCODE_GOOD:
        return 0;
      default:
  	    return 1;
    }

	/* deallocate certificate's memory */
	UA_Server_delete(pPriv->server);
	pPriv->nl.deleteMembers(&pPriv->nl);
}
```

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [x] Other OPC UA SDKs used (client or server):  UA Modeler
 - [x] Operating system: Ubuntu 14.04.5 LTS, cross compile 'cortexa9hf-neon-poky-linux-gnueabi '
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

lehned, 2018-01-24:
I may found the location of the actual situation. Initial question stays for discussion.

```
./src/server/ua_services_nodemanagement.c-871-    /* Instantiate variables and objects */
./src/server/ua_services_nodemanagement.c-872-    if(node->nodeClass == UA_NODECLASS_VARIABLE ||
./src/server/ua_services_nodemanagement.c-873-       node->nodeClass == UA_NODECLASS_OBJECT) {
./src/server/ua_services_nodemanagement.c-874-        UA_assert(type != NULL); /* see above */
./src/server/ua_services_nodemanagement.c-875-        /* Add (mandatory) child nodes from the type definition */
./src/server/ua_services_nodemanagement.c-876-        if (!server->bootstrapNS0) {
./src/server/ua_services_nodemanagement.c:877:            retval = addChildren(server, session, node, type);
./src/server/ua_services_nodemanagement.c-878-            if(retval != UA_STATUSCODE_GOOD) {
./src/server/ua_services_nodemanagement.c-879-                UA_LOG_INFO_SESSION(server->config.logger, session,
./src/server/ua_services_nodemanagement.c-880-                                    "AddNodes: Adding child nodes failed with error code %s",
./src/server/ua_services_nodemanagement.c-881-                                    UA_StatusCode_name(retval));
./src/server/ua_services_nodemanagement.c-882-                goto cleanup;
./src/server/ua_services_nodemanagement.c-883-            }
./src/server/ua_services_nodemanagement.c-884-        }
```

lehned, 2018-01-25:
Adding this patch makes a workaround but all checks (node collisions, type checks, ...) are disabled.

```
Index: open62541/include/ua_server.h
===================================================================
--- open62541.orig/include/ua_server.h	2018-01-03 12:51:43.476425115 +0000
+++ open62541/include/ua_server.h	2018-01-25 09:09:18.277861999 +0000
@@ -1048,6 +1048,11 @@
                                             *(UA_UInt64*)(uintptr_t)&jobId);
 }
 
+/** disableing any checks and type interpretation **/
+typedef UA_StatusCode (*UA_Model)(UA_Server *server);
+
+UA_StatusCode UA_Server_addModel(UA_Server *server, UA_Model pfModel);
+
 #ifdef __cplusplus
 }
 #endif
Index: open62541/src/server/ua_services_nodemanagement.c
===================================================================
--- open62541.orig/src/server/ua_services_nodemanagement.c	2017-12-22 11:56:29.606251897 +0000
+++ open62541/src/server/ua_services_nodemanagement.c	2018-01-25 09:10:30.253831999 +0000
@@ -5,6 +5,23 @@
 #include "ua_server_internal.h"
 #include "ua_services.h"
 
+/******************************/
+/* Add Model Bootstrap Active */
+/******************************/
+UA_StatusCode
+UA_Server_addModel(UA_Server *server, UA_Model pfModel)
+{
+  if (pfModel == (UA_Model)(0UL))
+    return UA_STATUSCODE_BADINTERNALERROR;
+
+  server->bootstrapNS0 = true;
+  UA_StatusCode retval = pfModel(server);
+  server->bootstrapNS0 = false;
+
+  return retval;
+}
+
+
 /*********************/
 /* Edit Node Context */
 /*********************/
```

Pro, 2018-01-26:
Thanks for your report!

I did the following test:

Just put your XML definition into `open62541/examples/nodeset/server_nodeset.xml` and then compile and run `server_nodeset`.

The output is as you mentioned:
[image]

The first `at` has node id 53532 and inherits EURange from the type `batype`.

The generated C code is here:
https://pastebin.com/tYxTpXWB

The steps are currently the following:
1. NS0 is created
2. ObjectType `batype` is added to the server with all its children
3. `myb` is added to the server and has type definition of `bytype`. Since `batype` has a `Mandatory` `at` child, it is added to `myb`. This `at` has then a node ID bigger than 50000 (all automatic nodes ids are in that range). Since it is modeled as mandatory, this behaviour is correct.
4. An additional `at` node with ID 6009 is modeled in the XML and therefore added to `myb`. The type of `at` also indicates the mandatory child `EURange` which is also added in this step.
5. The XML then defines an additional `EURange` (id=6010) variable as child of 6009

Therefore you get this output.

So there are two options:
1. UaModeller is generating the wrong model (i.e. it should not generate the mandatory nodes), but then it would not be possible to set specific values for an instance of a type.

2. We need to first create all the nodes of a specific XML, and at the end (after all nodes are added) we need a method which checks for all mandatory children, and if they are not there yet, they need to be created, instead of creating the mandatory children as soon as the node is added. This is only the case if the nodes are created from the nodeset compiler output. If a node is added from within the user source code, the mandatory children need to be created immediately. Or should we also allow the option that mandatory children are created after some calls?

Probably it's option 2 we should go for?

Pro, 2018-01-26:
Ah, I just remembered that the mandatory children are added in the `AddNode_finish` step. So we would need to change the Nodeset Compiler to use the corresponding `_begin` and `_finish` methods and only call `_finish` at the end when all children of a node are added, that should fix it.

lehned, 2018-01-31:
Hi, thanks for reply. 
Yes, a  possible solution is to split the parent nodes to _begin and finish_ and all leaf nodes as normal.
Semantic insert from XML described nodes should be:
```
addnode_begin (myb, OBJECT)
addnode_begin(at, VARIABLE)
addVariableNode(EURange)
addnode_finish(at, VARIABLE)
addnode_finish(myb, OBJECT)
```
Is this correct?

Pro, 2018-02-01:
Yes, that would be the correct way to use the methods. We would need to change the nodeset compiler to do the same.

jpfr, 2018-02-01:
> Yes, that would be the correct way to use the methods. We would need to change the nodeset compiler to do the same.

That would be rather easy. The nodes are sorted before printing.
- First pass: Print addnode_begin for every node
- Second pass: Go over the list in reverse order and print addnode_finish.

Pro, 2018-02-01:
I created a fix for this issue in #1568 

The model now looks like this which should be correct. Can you please check?
[image]

PR #1568 (2018-02-02): Use _begin and _finish methods in Nodeset Compiler
Fixes #1547 wrong inheritane of child nodes
commit 0dd0f444 (2018-02-02): Use _begin and _finish methods to ensure inherited children are not created multiple times

fixes #1547
closed 2018-02-02 by Pro