open62541/open62541 #1548: Connect to the selected server with the given username and password

mmesa2016, 2018-01-24:
My Ingles is bad.
Connect to server with the given usarname and password ( Service --> UA_Client_connect_username), return UA_STATUSCODE_BADINTERNALERROR ("No suitable endpoint found"). In service getEndpoints in file src/client/ua_client_connect.c (master branch) in line  309 --> if(!UA_String_equal(&endpoint->securityPolicyUri, &securityNone)). After line replace for  if(UA_String_equal(&endpoint->securityPolicyUri, &securityNone)). An line 320 - 321 --> if(userToken->securityPolicyUri.length > 0 &&
!UA_String_equal(&userToken->securityPolicyUri, &securityNone)). After line replace for if(userToken->securityPolicyUri.length > 0 &&
UA_String_equal(&userToken->securityPolicyUri, &securityNone)).

Pro, 2018-02-01:
This is probably the same as #1539

closed 2018-02-01 by Pro

Pro, 2018-02-01:
Also duplicate of #934