open62541/open62541 #1554: UaExpert client connect to server using BasicRsa128

forkmaster7, 2018-01-29:
!ATTENTION!
Please read the following page carefully and provide us with all the
information requested:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Use Github Markdown to format your text:
https://help.github.com/articles/basic-writing-and-formatting-syntax/

Fill out the sections and checklist below (add text at the end of each line).

Remove this text between the ATTENTION marks after reading it.
!ATTENTION!
--------------------------------------------------------------------------------

Description
===========
UaExpert client running on windows system does not connect to the open62541 server. Used the open 62541 server code (server_basic128rsa15) available in the example folder.

The error log in the terminal:
`info/network	Connection 5 | New connection over TCP from 192.168.1.96
[2018-01-28 13:28:43.416 (UTC+0550)] info/channel	Creating a new SecureChannel
[2018-01-28 13:28:43.423 (UTC+0550)] info/channel	Connection 5 | SecureChannel 4 | Opened SecureChannel
[2018-01-28 13:28:45.099 (UTC+0550)] warn/session	Connection 0 | SecureChannel 0 | Session 91f5f0fc-5281-1c9b-f51f-d13def3ef3bf | Calling service 473 on a non-activated session
[2018-01-28 13:28:45.099 (UTC+0550)] info/channel	Connection 5 | SecureChannel 4 | CloseSecureChannel
[2018-01-28 13:28:45.099 (UTC+0550)] info/network	Connection 5 | Closed by the server`

I have generated the certificate and key using the given tool. Additionally i have exported the .der file as .pem for the UaExpert client.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): current master branch 0.3
 - [x] Other OPC UA SDKs used (client or server): UaExpert Client
 - [x] Operating system: Ubuntu 16.04
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Infinity95, 2018-01-29:
I cannot reproduce your issue. Can you be a bit more specific about what steps need to be done to produce the issue?
You shouldn't need to copy anything to uaexpert. You *may* specify the uaexpert certificate as trusted when starting the server. The default will just acceppt all certificates, trusted or not.

forkmaster7, 2018-01-30:
@Infinity95 Thanks for the info.
I have also tested with default certificate in the UaExpert, it works.

I have the certificate generation tool (Path: tools/certs/create_self-signed.py) for generating the key. Additionally i have included `os.system("openssl rsa -inform PEM -in localhost.key -outform PEM -out server_key.pem")` in that python for .pem file which is required by the UaExpert.
I ran the server application with certificate and private key. And the same certificate and .pem key file used in the UaExpert client for connecting to the server.

 Please let me know if this is repeatable!

jpfr, 2018-01-30:
>  I have also tested with default certificate in the UaExpert, it works.

Great. That's the goal.

> And the same certificate and .pem key file used in the UaExpert client for connecting to the server.

This is not how public key encryption works.
You are not expected to load the private key of the server into UAExpert.
Just add the server certificate to the trust-list.
Or build up a certificate chain and add the root certificate to the trust-list.

At any rate, what is the log output in UAExpert?
You can activate the log window under `View->Log`. You can also write more verbose output to a log file via the settings.

forkmaster7, 2018-01-30:
@jpfr Thanks. This actually makes lot of sense to me.

Can you please help me with specific steps to repeat?

Should i use the server certificate alone in the UaExpert client, what type of file is that, server supports only .crl file for trust list addition

jpfr, 2018-01-30:
You can add the certificate alone.
On my box, this changes the messages from "Untrusted" to "Trusted but the Root Certificate is unknown".

@Infinity95 
Can we write a short paragraph on how to add the root certificate and the chain of trust to UAExpert?
Adding the ca.crt to the Issuer list was not enough to get rid of all warnings.

Infinity95, 2018-01-30:
I'm not quite sure why UAExpert behaves that way. I'll need to do some testing to find out how UAExpert verifies the certificate chain. Maybe we need to send the whole chain in the open response.

forkmaster7, 2018-01-30:
@jpfr @Infinity95 But in the client side we may have to use the `UA_SecurityPolicy_Basic128Rsa15` function for opening the secure channel. If so, then the client should also use the certificate.der and privatekey.der (the function parameters), not the trust list certificate right?

Please throw some light on this understanding.

jpfr, 2018-01-30:
`privatekey.der` is private to the server.
Not even the client can know it. Still, both sides can establish a shared secret session key that nobody else can know.

That is the beauty of public key encryption. Don't give out your private key. Ever.

Pro, 2018-02-01:
Open ActionItem before closing this issue:

> @Infinity95
> Can we write a short paragraph on how to add the root certificate and the chain of trust to UAExpert?
> Adding the ca.crt to the Issuer list was not enough to get rid of all warnings.

bobbthebuilder, 2018-02-02:
In accordance with @forkmaster7's i
1. Ran `tools/certs/create_self-signed.py` which generated four files `ca.crt`, `ca.key`, `server_cert.der` and `server_key.der`. 
2. Compiled `examples/encryption/server_basic128rsa15.c` and ran the executable: `./server path/to/server_cert.der path/to/server_key.der`.
3. Set Security Settings->Security Policy to **Basic128Rsa15**,
    Set Security Settings->Message Security Mode to **Sign&Encrypt**
    Set Authentification Settings to **Anonymous**
4. Opened UAExpert and clicked Connect Server that opens the Certificate Validation window that shows the error **BadCertificateChainIncomplete** with Trust Status Untrusted and the error section lists
unable to get local issuer certificate [BadCertificateChainIncomplete]
unable to get certificate CRL [BadCertificateRevocationUnknown]
unable to verify the first certificate [BadCertificateChainIncomplete]
Clicking "Trust Server Certificate and Continue let the client successfully connect with the server.
(The Trust Status states Trusted afterwards, opposed to "Trusted but the Root Certificate is unknown" as @jpfr experienced).

I was not yet able to get rid of the error messages.

Remaining Problems/Questions:

a. When I set Authentification Settings to **Username and Password** i cannot connect with the server, UAExperts Log states: **Error 'BadConfigurationError' was returned during ActivateSession**. What is missing?

c. After copying `ca.crt` to `.config/unifiedautomation/uaexpert/PKI/issuers/certs` the Manage Certificate window->Issuers indicates with Status **Known Issuer** indicates that the certificate is correctly recognized.
(@jpfr did you do the same?). Unfortunately, there's absolutely no visible effect or changes, the error messages are the same.

d. UAExperts supports Authentification by Certiticate and Private Key. I assume another Certificate is needed. But how to put that one to the server as well? How's that done in general?
 
Thank you for any hints and help! :-)

forkmaster7, 2018-02-03:
@jpfr @Infinity95 when i set the **authentication** setting to certificate and private key the server throws **BadCertificateInvalid**. 
Im actually trying with the symmetric way of encryption.

I have created a key pair on two systems, one with open62541 tool and the other with puttygen. And i have used the client side certificate in the server and the server certificate in the client side.

Any comment on this error!

Infinity95, 2018-02-08:
certificate authentication is currently not supported. It would require changes in the accesscontrol plugin.

jpfr, 2018-02-08:
@Infinity95 @forkmaster7 
The authentication token is sent as an extensionobject. So the access control plugin API is ready for any type of authentication.

In that sense, we already support it. But the user has to make his custom adaptations.
Which are required anyway since the default plugin is very permissive.

Infinity95, 2018-02-08:
It would be possible to supply a default implementation that handles encrypted passwords and certificates though, right? That could be used as a base for more complex systems that the user has to implement himself.

jpfr, 2018-09-01:
We changed the way self-signed certificates are generated.
Did this fix the issue for you?

jpfr, 2019-02-10:
Seems to be resolved.
Please reopen if required.

closed 2019-02-10 by jpfr