open62541/open62541 #1569: Fixup for encoding of overlayable types likes enums

Pro, 2018-02-01:
This issue is here to not forget about the still needed fix for overlayable enums as discussed in
#1559

jpfr, 2018-02-03:
Copied from #1559 

Hello @janitza-thbe ,

I am trying to resolve the topics you mentioned.

On my box, the enums are already correctly transmitted without this PR!
The Datatype of enums has the builtin flag set to true and the type-index of int32.
So they are treated exactly as int32 in the encoding layer.

What are the steps to reproduce the error in your initial screenshot?

janitza-thbe, 2018-02-05:
I've an adapted `generate_datatypes.py` that accepts multiple model `*.xml` to builtin our data types with configurable namespace indexes, too. As I understood these must not get the flag `builtin` set because they are 1. different from `ns=0` and 2. are no `Builtin` types due to the OPC UA terminology.

Maybe just only a naming confusion but I think my originating idea in #1559 to introduce a `kind` field to easily differentiate the data types makes still sense and should be implemented before 0.3 release because it extends the API.

jpfr, 2020-07-07:
#1559 was merged long ago.

closed 2020-07-07 by jpfr