open62541/open62541 #1575: Bad node hash implementation

janitza-thbe, 2018-02-02:
Description
===========
The node hash, that is defined by `UA_NodeId_hash()` in `ua_types.c` is very bad and results in hash collisions for nearly every node that is added to namespaces with index `ns>0` and sequential numeric IDs with begin `i=1`. This unravels the efficiency of hash based implementations.
(The problem was detected by old nodestore which was based upon `liburcu` required even unique hashes!)

Original calculation:
```
// shift knuth multiplication to use highest 32 bits and after addition make sure we don't have an integer overflow
return (u32)((n->namespaceIndex + ((n->identifier.numeric * (u64)2654435761) >> (32))) & UINT32_C(4294967295)); /*  Knuth's multiplicative hashing */
```
As you can see the namespace index `ns` is just added to the number that was generated from numeric ID `i`. In case of nearly same numeric ID's and low namespace indexes `ns` this collides very often.

In my case I replaced the calculation by the following assignation that is much more better for this cases (indeed it even generates unique hashes for `0 < i < (1<<24)`) but does not generally solve the problem to get a good `uint32_t` hash of a random `uint16_t/uint32_t` pair.

```
return ((u32)((u32) n->namespaceIndex) << 24) + ((u32)(n->identifier.numeric & 0xffffff)); /* Bytes (MSB->LSB): (NS-Byte1), (NodeId-Byte3), (NodeId-Byte2), (NodeId-Byte1)  */
```

Checklist
=========
Please provide the following information:

 - [X] open62541 Version (release number or git tag): master

jpfr, 2018-02-02:
> In case of nearly same numeric ID's and low namespace indexes ns this collides very often.

The numeric id is multiplied with a very large prime number.
So similar ids do not lead to similar hashes.

This is a serious issue. And a fickle one where problems are not obvious at first glance.
I'm sure you found out about it via some measurements.
Can you share the numbers and/or measurement setup?

janitza-thbe, 2018-02-05:
Indeed I misinterpreted the result and didn't investigate much more after the first collisions occurred and broke my application start. The time I "fixed" it for myself is a long time ago to work with the old nodestore that used `liburcu`.

But generally this occured on loading a big model as `ns=1` that was setup using UAModeler with ascending node IDs up to a few thousand (`1 < i < 10000`) and dynamic instantiation of much more objects at runtime.

A few minutes ago I reverted to the old hash function that is still used by you and my application crashed again at the dynamic instantiation of our main object that holds hundreds of nested children (the detailed error messages come from my variant since the log messages were useless before w/o node IDs):

```
[2018/02/05 07:36:04.430] error/app	(Server.cpp:502) Failed to insert node ns=1;i=53893/@Q\�� into node map!
17: [2018/02/05 07:36:04.430] error/sess	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Global pre-constructor call for node "[ns=1;i=53893]" failed with code BadInternalError
18: [2018/02/05 07:36:04.430] warn/sess	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Copying child node "[ns=1;i=20]" of node "[ns=1;i=1]" to node "[ns=1;i=53892]" failed with code BadInternalError
19: [2018/02/05 07:36:04.430] warn/sess	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Failed to copy children of node "[ns=1;i=1]" to node "[ns=1;i=53892]" with code BadInternalError
20: [2018/02/05 07:36:04.430] error/sess	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Adding child nodes for node "[ns=1;i=53892]" failed with error code BadInternalError
```

This shows that the new nodestore plugin also requires unique node ID hashes which is generally wrong and even a **bug**! Every hash based implementation has to handle equal hashes (by loosing some performance).

janitza-thbe, 2018-02-05:
Indeed I misinterpreted the result and didn't investigate much more after the first collisions occurred and broke my application start. The time I "fixed" it for myself is a long time ago to work with the old nodestore that used the knuth's multiplicative _hashes as unique keys_ for the node map which lead to collisions at least for `ns=1;i=53893` with a node from full NS0.
This was high-grade wrong since hashes may be non-unique but regular maps require unique keys.

How the current nodestore works I didn't investigate further since I've unique hashes now that should never cause any problems under the specified range conditions and cannot test it at moment with the old algorithm.

jpfr, 2018-02-05:
> old nodestore that used the knuth's multiplicative hashes as unique keys
> How the current nodestore works I didn't investigate further since I've unique hashes now

The nodestores (old and new) never assumed that hashes are unique.
We always test for equality of the nodeid in addition to the hash.

Why do you need unique hashes when every node already has a unique nodeid?
Lookup with the nodeid is really fast. On the order of 300 processor cycles on my box.

I'm closing this. We can redo the hashing when we have a performance problem due to collisions.

closed 2018-02-05 by jpfr