open62541/open62541 #1580: Can´t run the example server implementation from the Readme.MD

weskampj, 2018-02-05:
## Can´t run the example server implementation from the Readme.MD

I tried to follow the small example from the readme but i run into some errors. When executing the command " run gcc -std=c99 <server.c> open62541.c -o server" i get the errors:

` ` `
open62541.c: In function 'socket_set_nonblocking':
open62541.c:37396:48: error: 'NO_ERROR' undeclared (first use in this function)
     if(ioctlsocket(sockfd, FIONBIO, &iMode) != NO_ERROR)
                                                ^~~~~~~~
open62541.c:37396:48: note: each undeclared identifier is reported only once for each function it appears in
open62541.c: In function 'socket_set_blocking':
open62541.c:37414:48: error: 'NO_ERROR' undeclared (first use in this function)
     if(ioctlsocket(sockfd, FIONBIO, &iMode) != NO_ERROR)
                                                ^~~~~~~~
open62541.c: In function 'addServerSocket':
open62541.c:37550:42: error: 'IPV6_V6ONLY' undeclared (first use in this function)
        setsockopt(newsock, IPPROTO_IPV6, IPV6_V6ONLY,
                                          ^~~~~~~~~~~
open62541.c: In function 'UA_DateTime_localTimeUtcOffset':
open62541.c:38166:5: warning: implicit declaration of function 'gmtime_s' [-Wimplicit-function-declaration]
     gmtime_s(&ptm, &rawtime);
     ^~~~~~~~
` ` `

Afterwards i tried to run the example with Visual Studio 2015 Version 14.0.25431.01 Update 3 with x86 debug configuration . Here i got the following Linker Tools Errors:

Fehler	LNK2019	Verweis auf nicht aufgelöstes externes Symbol "_UA_Variant_setScalar" in Funktion "_main".	Projekt1	\\fs1\Users\wes\Desktop\Tut_VS\Projekt1\myServer.obj	1	
Fehler	LNK2019	Verweis auf nicht aufgelöstes externes Symbol "_UA_Server_new" in Funktion "_main".	Projekt1	\\fs1\Users\wes\Desktop\Tut_VS\Projekt1\myServer.obj	1	
Fehler	LNK2019	Verweis auf nicht aufgelöstes externes Symbol "_UA_Server_delete" in Funktion "_main".	Projekt1	\\fs1\Users\wes\Desktop\Tut_VS\Projekt1\myServer.obj	1	
Fehler	LNK2019	Verweis auf nicht aufgelöstes externes Symbol "_UA_Server_run" in Funktion "_main".	Projekt1	\\fs1\Users\wes\Desktop\Tut_VS\Projekt1\myServer.obj	1	
Fehler	LNK2019	Verweis auf nicht aufgelöstes externes Symbol "___UA_Server_addNode" in Funktion "_UA_Server_addVariableNode".	Projekt1	\\fs1\Users\wes\Desktop\Tut_VS\Projekt1\myServer.obj	1	
Fehler	LNK2019	Verweis auf nicht aufgelöstes externes Symbol "_UA_ServerConfig_new_minimal" in Funktion "_UA_ServerConfig_new_default".	Projekt1	\\fs1\Users\wes\Desktop\Tut_VS\Projekt1\myServer.obj	1	
Fehler	LNK2019	Verweis auf nicht aufgelöstes externes Symbol "_UA_ServerConfig_delete" in Funktion "_main".	Projekt1	\\fs1\Users\wes\Desktop\Tut_VS\Projekt1\myServer.obj	1	
Fehler	LNK2001	Nicht aufgelöstes externes Symbol "_UA_NODEID_NULL".	Projekt1	\\fs1\Users\wes\Desktop\Tut_VS\Projekt1\myServer.obj	1	
Fehler	LNK2001	Nicht aufgelöstes externes Symbol "_UA_TYPES".	Projekt1	\\fs1\Users\wes\Desktop\Tut_VS\Projekt1\myServer.obj	1	
Fehler	LNK2001	Nicht aufgelöstes externes Symbol "_UA_VariableAttributes_default".	Projekt1	\\fs1\Users\wes\Desktop\Tut_VS\Projekt1\myServer.obj	1	

I just want to have an running example, which i can use to play around and getting some insights. I´m totally new to OPC UA. So thank you in advance for your help

Background Information / Reproduction Steps
===========================================
For reproducing the first error, just follow the example from readme.md. I am using Windows 10 Pro Version 1703 and the open62541 files with git tag "09c0d759af".
For the error in VS just create an empty project, add the header and the c file, create the server.c with the sample code from the readme.md and run it.

Checklist
=========
Please provide the following information:

 - [09c0d759af ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10 Pro Version 1703
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Stasik0, 2018-02-05:
run gcc -std=c99 <server.c> open62541.c -o server -lws2_32

> Am 05.02.2018 um 09:04 schrieb weskampj <notifications@github.com>:
> 
> run gcc -std=c99 <server.c> open62541.c -o server

weskampj, 2018-02-05:
> run gcc -std=c99 <server.c> open62541.c -o server -lws2_32
i tried this too. Same problem

Pro, 2018-02-05:
You are mixing two different problems.

Problem 1 (gcc on Windows):
It looks like you are using gcc on windows. Which package (mingw, cygwin, ...) are you using for that?
The `NO_ERROR` is a windows-specific define, as it is within an `#ifdef _WIN32`, so it could be that your compiler is setting the `_WIN32` define but not including the correct files.
Can you post your specific setup so we can investigate that further? Maybe there is a problem with mixing different defines for detecting windows within the source (i.e. `__MINGW32__`, `_MSC_VER`, `_WIN32`)

Problem 2 (Visual Studio):
The VS project is probably not set up correctly. My guess, you are only compiling the `example.c`, but not together with the `open62541.c` file.

weskampj, 2018-02-05:
I´m using MinGW. I´m not completly sure which information you need but in the mingw installation manager i choosed:

mingw-developer-toolkit 
mingw32-base
mingw32-gcc-g++
mysys-base

I have a 64 bit system. What else do you need?

ccvca, 2018-02-05:
> `error: 'IPV6_V6ONLY' undeclared`
I ran into this problem, too, when I used MinGW. My solution was, to use [mingw-w64](https://sourceforge.net/projects/mingw-w64/) which is much newer (gcc-7 instead of gcc-5).

Pro, 2018-02-05:
I remember that there were some problems using the "normal" mingw, also on AppVeyor.

Can you try to uninstall your version, and then use this one:
https://sourceforge.net/projects/mingw-w64/files/Toolchains%20targetting%20Win32/Personal%20Builds/mingw-builds/installer/mingw-w64-install.exe/download

Appveyor is also using mingw-w64 and thus compiling the stack and examples using that should work. See
https://github.com/open62541/open62541/blob/master/appveyor.yml#L53

weskampj, 2018-02-05:
Great now i was able to create an executable. I used the MinGw-w64 and the "gcc -std=c99 open62541.c myServer.c -o myServer -lws2_32" command. Without the "lws2_32 i got many undefined references errors as in VS. Thank you

Regarding VS you have an idea? Both files the open62541.c and the server.c are complied successfully. Should it generally work with Visual Studio to use, build and debug the library?  It would be awesome if i can debug and run Open62541 with VS but if not it is okay ;)

Pro, 2018-02-05:
Yes, visual studio should also be able to compile everything together. It should be enough to add both files to the project.
See also:
https://stackoverflow.com/questions/12074696/adding-existing-h-and-c-files-to-a-microsoft-visual-studio-2010-c-project

closed 2018-02-05 by Pro

ccvca, 2018-02-05:
Do you add `ws2_32.lib;` as an additional linker input?

weskampj, 2018-02-06:
For your interest. I was able to build the readme.md example with Visual Studio after adding the ws2_32.lib to the project settings. You can do this under Configuration properties - Linker - Input - Additional Dependencies