open62541/open62541 #1584: [Server][Feature Request] Callback to application when client tries to read non-existing node

JoJoBond, 2018-02-06:
Description
===========
I would like to ask for a feature where the server triggers a application callback whenever a client tries to read/subscribe to a node. The application can react to that request, e.g. create the node ad hoc, alter/map the nodeid to a different one, etc.

Background Information / Reproduction Steps
===========================================
Often, especially when you are working with PLCs, you don't know what data is available on your source side. So a client must 'blindly' read a non-browseable value and the server will try to provide that value based on predefined node syntax. When using OPC DA servers for accessing Siemens S7 PLC data you are often able to directly address the data that you need, e.g.
"DB10.Real16" reads a 4 byte real on byte position 16 in data block 10.
The server will read that data from PLC and relay it to the client. After reading the node once, it becomes browseable and can be used as usual afterwards.

I don't have working cmake environment but did a very basic implementation on the single file v0.2 sources:

open62541.c: https://paste2.org/nLf7J3m7
open62541.h: https://paste2.org/AemjW7ax
patch file: https://paste2.org/L1yk7BKb

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Does not apply
 - [ ] Other OPC UA SDKs used (client or server): Does not apply
 - [ ] Operating system: Does not apply
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: Does not apply
 - [ ] Wireshark network dump attached: Does not apply
 - [ ] Self-contained code example attached: Does not apply
 - [ ] Critical issue: No

StalderT, 2018-02-06:
Good idea. I will propose a PR.

jpfr, 2018-02-06:
There are two possibilities:

- Add the callback to the server that is used in the read service
- Add the callback to the nodestore plugin (where nodestore_get fails to find the node)

I tend to minimize the API "surface" of the main library.
The nodestore is a "plugin" and free for modification.
What are the optinions from @open62541/open62541-core ?

By the way, **if you modify the nodestore plugin, you can return the created node already for the first read.** So you don't have to "read twice" to get data out.

StalderT, 2018-02-06:
If you want : https://github.com/StalderT/open62541/commit/b3c015ce7295c6e29ade6b0de182dd3592a72f8b

> By the way, if you modify the nodestore plugin, you can create the missing node already for the first read. So you don't have to "read twice" to get data out.

With this commit, the node is created at the first read

StalderT, 2018-02-07:
Most of the opc servers I've been able to test (in industrial automation) have the functionality to create a node when a client tries to access it. This in order to access a PLC variable.

Adding this feature as an API as it is proposed here would be, I think, a good feature. This would easily develop an opc server with this feature.

jonasgreen88, 2018-02-08:
I agree to @jpfr that this is rather a nodestore feature than a stack feature. If a nodestore receives a request of node that doesn't exist, the nodestore is free to create it. The stack doesn't bother if the node existed before the request or if it was just created. The feature could be implemented in the default node store plugin implementation to show an example.

JoJoBond, 2018-02-09:
@jpfr @jonasgreen88 
How does the following look?

https://github.com/JoJoBond/open62541/commit/7c4703bddf36a5230a62942125bf45b27eb32ba5

jpfr, 2018-02-15:
@JoJoBond Looks cool!
Are you okay if we leave it like that?
The nodestore is intended to be modified for specific use.
It seems the feature can be implemented without having to changing any non-plugin-specific API.

JoJoBond, 2018-02-15:
@jpfr 
Yes sure, I'm quite happy with this solution.
Nodestore really is the right place to implement such a feature.
If you wish to change any of the function or variable naming go ahead.

Root-Core, 2018-03-17:
@JoJoBond This is quiet useful, as it allows us to check dynamic client configurations. Nice work.
Could you open a pull request to get this merged?

JoJoBond, 2018-03-17:
@Root-Core 
You should be able to set the handlerContext even after passing the config.
UA_ServerConfig_new_default calls UA_ServerConfig_new_minimal.
There UA_Nodestore_default_new is called to create the nodestore which is stored in conf->nodestore. The context (actually the NodeMap) that is required for UA_NodeMap_setNotFoundCallback is stored in nodestore->context.
So, after calling UA_ServerConfig_new_default, you can store the pointer value (address) of conf->nodestore->context in a separate pointer and use it to enable/disable the callbacks even while the server is running. That way you can pass the server pointer as handlerContext.

I will do some clean up of my code (naming/indentation) and will open a pull request once I'm finished.

Pro, 2018-11-06:
Closing as wontfix since the nodestore should be adapted for such specific use-cases.

closed 2018-11-06 by Pro