open62541/open62541 #1592: Unable to download the Single file distribution.

SenthilKumarGS, 2018-02-13:
When i try to download the single file distributions from "https://github.com/open62541/open62541/releases/download/v0.2/open62541.c" using 'wget' or 'curl', i get the following error.
"Unable to establish SSL connection."

This has to be because of the firewall settings in my company network.

I am able to download it from https://open62541.org/releases/ but those are for testing purposes and may be removed without notice. 
I am writing a script(using SCons) to automate the downloading and building of this library. 
If i take one of the version from https://open62541.org/releases/, it may not work after some days when the version is removed from the link. So i can't use them.

Is there any other link where single file distributions are kept forever?

Any help/suggestion will be highly appreciable. Thanks.

ccvca, 2018-02-13:
This looks more like a github.com issue than an issue with open62541.
Can you download any file from github.com with wget?

SenthilKumarGS, 2018-02-13:
I am unable to download any file from github.com because of firewall i guess.
There is no issue with open62541. My actual concern is on the availability single file distribution.

SenthilKumarGS, 2018-02-13:
I have the following doubts.
Is single file distribution for v0.2 available?
Will there be a single file distribution release for each and every commit in master branch? 
Are there any links where single file distributions are kept forever?

SenthilKumarGS, 2018-02-13:
Found the single file distribution for v0.2 @ https://github.com/open62541/open62541/releases
I am not able to view/download them due to company firewall.

I was looking for a way to download single-file distribution bypassing firewall but ended-up with a different approach which is cloning the latest master branch code and building it for single-file distribution using 'UA_ENABLE_AMALGAMATION' build flag.

closed 2018-02-13 by SenthilKumarGS

Pro, 2018-02-14:
Depending on your use-case it's probably best to not use master, but some specific branch like 0.2 or (soon to be coming) 0.3. Master is not stable