open62541/open62541 #1615: Writing values in XML server

tbobek, 2018-02-25:
Description
===========

This is rather a question how to approach the XML server functionality than an issue with the code. But after scanning through all the documentation, searching the internet and a lot of modify/compile/run cycles I want to ask you - the experts - for help. I'm using Ubuntu Linux 17.10, gcc 7.2.0, cmake 3.9.1 and checked out `master` yesterday. Compiling the sources worked so far, together with examples and nodeset generation. My goal is to set up an XML based OPC-UA server and writing simple basic type values into the nodes of the XML description. 

As a first attempt, I extended the example `server_nodeset` with code that should write a value to the double entry (ns=1, i=6001). Following the example in the tutorials, I used the [write service](https://open62541.org/doc/current/tutorial_server_variable.html). This is the point where I got stuck with the XML setup. 

Background Information / Reproduction Steps
===========================================

My modified `server_nodeset`example looks like this 
```
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#include <signal.h>
#include "open62541.h"
#include <stdio.h>
/* Files example_namespace.h and example_namespace.c are created from server_nodeset.xml in the
 * /src_generated directory by CMake */
#include "example_nodeset.h"

UA_Boolean running = true;

static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

static void writeVariable(UA_Server *server) {
    UA_NodeId myDoubleNodeId = UA_NODEID_NUMERIC(1, 6001);

    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "nodeid: %d, %s", myDoubleNodeId.identifier.numeric, myDoubleNodeId.identifier.string.data);
    UA_Double myDouble = 99.987;
    UA_Variant myVar;
    UA_Variant_init(&myVar);
    UA_Variant_setScalar(&myVar, &myDouble, &UA_TYPES[UA_TYPES_DOUBLE]);

    UA_StatusCode result = UA_Server_writeValue(server, myDoubleNodeId, myVar);
    if( result != UA_STATUSCODE_GOOD) {
      UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "Could not write value: %8x", result);
    }
}

int main(int argc, char** argv) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server *server = UA_Server_new(config);

    UA_StatusCode retval;
    /* create nodes from nodeset */
    if(example_nodeset(server) != UA_STATUSCODE_GOOD) {
        UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "Could not add the example nodeset. "
        "Check previous output for any error.");
        retval = UA_STATUSCODE_BADUNEXPECTEDERROR;
    } else {

        writeVariable(server);

        retval = UA_Server_run(server, &running);
    }
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}
```

The changes are the `writeVariable` method and its call in  `main` . 
The `server_nodeset.xml` is used without changes. The server starts, calls the `writeVariable` method but then cannot write into the node because the nodeID is not correct. 

The output after starting `server_nodeset` looks like this: 

```
[2018-02-25 11:06:53.824 (UTC+0100)] info/server	nodeid: 6001, P��3�
[2018-02-25 11:06:53.824 (UTC+0100)] error/server	Could not write value: 80340000
[2018-02-25 11:06:53.825 (UTC+0100)] info/network	TCP network layer listening on opc.tcp://deepthought2:4840/
```

The code 0x80340000 is a bad node id. I modified also the ID values to match other entries in the XML tree, unfortunately without results. It's not clear to me if this approach is in general suitable to handle XML models or not. I would appreciate any help on this! 

### A second question
Is there also a recommendation that you could make in respect to **OPCUA-modeling software**. The free python based [opcua-modeler](https://github.com/FreeOpcUa/opcua-modeler) modifies the XML tree completely. The produced output is not usable for inserting in open62541 (to my first experience). 

Tanks in advance for your help! 

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux 17.10, gcc, cmake (see text)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

Stasik0, 2018-02-25:
Can you verify using uaexpert that the id of the variable is (1,6001)? There might be an additional namespace id in between.

tbobek, 2018-02-25:
Yes, funnily in UAExpert / Win10 the XML tree looks different. After switching to Win10 with mingw32, changing ID and data type (final method see below)  I get now positive feedback. UAExpert seems to be better fitting to open62541. 

Thanks for your help!

```
static void writeVariable(UA_Server *server) {
    UA_NodeId myUIntNodeId = UA_NODEID_NUMERIC(3, 6002);

    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "nodeid: %d", myUIntNodeId.identifier.numeric);
    UA_UInt32 myUInt = 21;
    UA_Variant myVar;
    UA_Variant_init(&myVar);
    UA_Variant_setScalar(&myVar, &myUInt, &UA_TYPES[UA_TYPES_UINT32]);

    UA_StatusCode result = UA_Server_writeValue(server, myUIntNodeId, myVar);
    if( result != UA_STATUSCODE_GOOD) {
      UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "Could not write value: %8x", result);
    }
   else {
		UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "node value %d was written", myUInt ); 
  }
}
```

Stasik0, 2018-02-25:
Namespace ids in xml files are *file-local*, since server may load multiple xml filed. The only way to find the runtime-id of a namespace is to match the namespace uri from the xml file with server's namespace array.

tbobek, 2018-02-25:
Thanks for the hint! I've added a method in `ua_server.c` to find the namespace number from a substring of the namespace url string. Now it becomes more convenient. With the Unified Automation Modeler and the UAExpert Client it works more reliable to test the nodes added to the XML.