open62541/open62541 #1616: `mbedtls_sha1` is deprecated

Pro, 2018-02-25:
Compiling with mbedtls 2.7.0 `mbedtls_sha1` is marked as deprecated:

```
I  ERROR: build step 4 "gcr.io/oss-fuzz/open62541" failed: exit status 2 
I  ERROR 
I  Finished Step #4 
I  Step #4: make: *** [all] Error 2 
I  Step #4: Makefile:149: recipe for target 'all' failed 
I  Step #4: CMakeFiles/Makefile2:135: recipe for target 'CMakeFiles/open62541-plugins.dir/all' failed 
I  Step #4: make[1]: *** [CMakeFiles/open62541-plugins.dir/all] Error 2 
I  Step #4: make[2]: *** [CMakeFiles/open62541-plugins.dir/plugins/ua_securitypolicy_basic128rsa15.c.o] Error 1 
I  Step #4: CMakeFiles/open62541-plugins.dir/build.make:270: recipe for target 'CMakeFiles/open62541-plugins.dir/plugins/ua_securitypolicy_basic128rsa15.c.o' failed 
I  Step #4: 3 errors generated. 
I  Step #4:                                                ^ 
I  Step #4: #define MBEDTLS_DEPRECATED      __attribute__((deprecated)) 
I  Step #4: /usr/include/mbedtls/sha1.h:302:48: note: expanded from macro 'MBEDTLS_DEPRECATED' 
I  Step #4: ^ 
I  Step #4: MBEDTLS_DEPRECATED static inline void mbedtls_sha1( const unsigned char *input, 
I  Step #4: /usr/include/mbedtls/sha1.h:320:1: note: 'mbedtls_sha1' has been explicitly marked deprecated here 
I  Step #4:     ^ 
I  Step #4:     mbedtls_sha1(certificate->data, certificate->length, thumbprint->data); 
I  Step #4: /src/open62541/plugins/ua_securitypolicy_basic128rsa15.c:265:5: error: 'mbedtls_sha1' is deprecated [-Werror,-Wdeprecated-declarations] 
I  Step #4:                                                ^ 
I  Step #4: #define MBEDTLS_DEPRECATED      __attribute__((deprecated)) 
I  Step #4: /usr/include/mbedtls/sha1.h:302:48: note: expanded from macro 'MBEDTLS_DEPRECATED' 
I  Step #4: ^ 
I  Step #4: MBEDTLS_DEPRECATED static inline void mbedtls_sha1( const unsigned char *input, 
I  Step #4: /usr/include/mbedtls/sha1.h:320:1: note: 'mbedtls_sha1' has been explicitly marked deprecated here 
I  Step #4:     ^ 
I  Step #4:     mbedtls_sha1(message->data, message->length, hash); 
I  Step #4: /src/open62541/plugins/ua_securitypolicy_basic128rsa15.c:113:5: error: 'mbedtls_sha1' is deprecated [-Werror,-Wdeprecated-declarations] 
I  Step #4:                                                ^ 
I  Step #4: #define MBEDTLS_DEPRECATED      __attribute__((deprecated)) 
I  Step #4: /usr/include/mbedtls/sha1.h:302:48: note: expanded from macro 'MBEDTLS_DEPRECATED' 
I  Step #4: ^ 
I  Step #4: MBEDTLS_DEPRECATED static inline void mbedtls_sha1( const unsigned char *input, 
I  Step #4: /usr/include/mbedtls/sha1.h:320:1: note: 'mbedtls_sha1' has been explicitly marked deprecated here 
I  Step #4:     ^ 
I  Step #4:     mbedtls_sha1(message->data, message->length, hash); 
I  Step #4: /src/open62541/plugins/ua_securitypolicy_basic128rsa15.c:90:5: error: 'mbedtls_sha1' is deprecated [-Werror,-Wdeprecated-declarations] 
I  Step #4: [ 14%] Building C object CMakeFiles/open62541-plugins.dir/plugins/ua_securitypolicy_basic128rsa15.c.o 
```

See also:
https://github.com/google/oss-fuzz/issues/1192#issuecomment-368297514

@Infinity95 can you have a look at that when you have some time. Thanks

Infinity95, 2018-02-27:
@Pro Can you check if the issue is fixed? The commit is now merged to master

aDogCalledSpot, 2018-03-14:
I discussed this with @Pro and we noticed the issue still wasn't fixed because it is still using deprecated functions.  Could you have another look at it? I linked a PR I made which caused these issues again below.

https://ci.appveyor.com/project/open62541/open62541/build/5771/job/qa8sfqg2texyla4f#L734

closed 2018-03-15 by jpfr