open62541/open62541 #1638: Developer wanted: Generate datatype according to the selected features

jpfr, 2018-03-07:
The datatype definitions generate some binary overhead.

The file `/tools/schema/datatypes_minimal.txt` is no longer that minimal, as it supports the types for all services supported by open62541,

Since some services can be selected in the cmake environment, we can be smart about it and only generate the required types. The following changes are required for this:

- Split the datatypes_minimal.txt into several per-feature files
- Adjust the script `/tools/generate_datatypes.py` to take several type-list files as arguments
- Adjust the `/CMakelists.txt` to use the appropriate type-lists based on the selected features

This issue is up for grabs. Please give short notice if you are working on it. So we can avoid duplicate efforts.

Infinity95, 2018-03-08:
Do we need the datatypes_*.txt files? Couldn't we parse the necessary information from the source code? That way we avoid having to maintain those txt files if something changes.
Since everything is prefixed by UA_ this should make things easier, right?

jpfr, 2018-03-09:
Would this require the parsing to know and understand the defined compiler macros?
The features are not always neatly in separate files.

I'm all for automation if it saves time. Not sure how much effort this will be though.

How about starting quick and dirty with datatypes_*.txt files and then see how much effort this is to maintain.

Stasik0, 2018-03-29:
https://www.hanselman.com/blog/XmlPreprocessIsAnElegantHackOrTheHackIsInTheEyeOfTheBeholder.aspx
Maybe something similar for txt files 
And a python-written preprocessor

Pro, 2018-07-11:
This was implemented in #1757

closed 2018-07-11 by Pro