open62541/open62541 #1639: Reason for forced static CRT

the-om, 2018-03-07:
Description
===========
Currently, when building open62541 as a static library with MSVC, the compiler flag /MD (link with dynamic CRT) gets replaced with /MT (static CRT) in CMakeLists.txt. This causes issues during linking if conflicting settings are used in other modules. I just would like to know if there is a good reason for forcing this setting and overriding the user's choice.

Background Information / Reproduction Steps
===========================================
Build mbedtls with dynamic crt
Build open62541 with msvc with encryption enabled

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Stasik0, 2018-03-07:
https://github.com/open62541/open62541/commit/b86938501a48e5c6c003b66df9d11e9d5d261582

Stasik0, 2018-03-07:
Maybe you could propose some override logic in the cmake file?

the-om, 2018-03-08:
Thank you for the link. I had seen that commit but somehow missed the comment.
What would be the best way to do this? Introduce a switch UA_FORCE_STATIC_CRT that's on by default?

Stasik0, 2018-03-08:
Yes, like UA_MSVS_FORCE_STATIC_CRT. Advanced and on by default. And then wrap the foreach loop in an if? @jpfr?

closed 2018-03-09 by the-om