open62541/open62541 #1643: Can not build with encryption and amalgamation

the-om, 2018-03-09:
Description
===========
#1634 breaks my build.
The reason are 3 violations of the one definition rule when using amalgamation.
The functions sha1, asym_getRemotePlainTextBlockSize and asym_getRemoteEncryptionKeyLength are defined as static, but when amalgamated they are defined twice in the same translation unit (open62541.c) which is not legal.

Background Information / Reproduction Steps
===========================================
Build with UA_ENABLE_AMALGAMATION and UA_ENABLE_ENCRYPTION

Compiler: Microsoft Visual Studio 2017

ccvca, 2018-03-09:
Possible solutions might be:
- Use an "include Guard" around the static function definitions,
- Use normal header/source and use only one implementation

jpfr, 2018-03-09:
- pull out the common code in common .c file that is used for both SecurityPolicy implementations
- rename the duplicate methods to something more specific

closed 2018-03-10 by the-om