open62541/open62541 #1644: Basic128Rsa15 Server with node-opcua client cannot create server nonce

thedraft, 2018-03-09:
Description
===========
I'm running the server_basic128rsa15 example with self signed certificates from /tools/cert/ and try to connect from a node-opcua (node-red-contrib-iiot-opcua) client with Security Policy Basic128Rsa15 and Security Mode Sign&Encrypt. The server creates a secure channel but cannot activate the session, so the client responds with "Error: Session Not Ready To Browse".

Server Log
===========================================
```
./server_basic128rsa15 server_cert.der server_key.der 
[2018-03-09 09:10:33.806 (UTC+0000)] warn/userland	No CA trust-list provided. Any remote certificate will be accepted.
[2018-03-09 09:10:34.513 (UTC+0000)] info/network	TCP network layer listening on opc.tcp://raspberrypi:4840/
[2018-03-09 09:19:14.649 (UTC+0000)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2018-03-09 09:19:14.650 (UTC+0000)] info/network	Connection 6 | New connection over TCP from 127.0.0.1
[2018-03-09 09:19:14.694 (UTC+0000)] info/channel	Creating a new SecureChannel
[2018-03-09 09:19:14.694 (UTC+0000)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
[2018-03-09 09:19:14.695 (UTC+0000)] info/channel	Connection 5 | SecureChannel 1 | Opened SecureChannel
[2018-03-09 09:19:14.741 (UTC+0000)] info/channel	Creating a new SecureChannel
[2018-03-09 09:19:14.742 (UTC+0000)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
[2018-03-09 09:19:14.742 (UTC+0000)] info/channel	Connection 6 | SecureChannel 2 | Opened SecureChannel
[2018-03-09 09:19:14.824 (UTC+0000)] info/channel	Connection 5 | SecureChannel 1 | CloseSecureChannel
[2018-03-09 09:19:14.825 (UTC+0000)] info/network	Connection 5 | Closed by the server
[2018-03-09 09:19:14.912 (UTC+0000)] info/channel	Connection 6 | SecureChannel 2 | CloseSecureChannel
[2018-03-09 09:19:14.912 (UTC+0000)] info/network	Connection 6 | Closed by the server
[2018-03-09 09:19:14.915 (UTC+0000)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2018-03-09 09:19:15.540 (UTC+0000)] info/channel	Creating a new SecureChannel
[2018-03-09 09:19:15.626 (UTC+0000)] info/channel	Connection 5 | SecureChannel 3 | Opened SecureChannel
[2018-03-09 09:19:17.008 (UTC+0000)] info/session	Connection 5 | SecureChannel 3 | Session d1d34fc9-732d-29d6-8558-65d6f75f6536 | ActivateSession: Could not generate a server nonce
[2018-03-09 09:19:17.016 (UTC+0000)] info/session	Connection 5 | SecureChannel 3 | Session 5d27d311-be08-c83d-f2bf-3b133403d04d | ActivateSession: Could not generate a server nonce
[2018-03-09 09:19:17.058 (UTC+0000)] info/session	Connection 5 | SecureChannel 3 | Session 555605d0-81d0-7f27-5ed5-4d14a51a3183 | ActivateSession: Could not generate a server nonce
[2018-03-09 09:20:23.847 (UTC+0000)] info/session	Connection 5 | SecureChannel 3 | Session 555605d0-81d0-7f27-5ed5-4d14a51a3183 | Session has timed out
[2018-03-09 09:20:23.847 (UTC+0000)] info/session	Connection 5 | SecureChannel 3 | Session 5d27d311-be08-c83d-f2bf-3b133403d04d | Session has timed out
[2018-03-09 09:20:23.848 (UTC+0000)] info/session	Connection 5 | SecureChannel 3 | Session d1d34fc9-732d-29d6-8558-65d6f75f6536 | Session has timed out
```

Additional Information
=========
open62541 Version: master\bee9016 as server
Other OPC UA SDKs used (client or server): node-opcua as client
Operating system: Raspbian

ChHaa, 2018-03-09:
What kind of authentication do you want to use for the session ? Note that securechannel and session are two different things.

thedraft, 2018-03-09:
I don't use login credentials, so I'm trying to open an anonymous session.

ChHaa, 2018-03-09:
I just checked out current master and tried with UA Expert:

[2018-03-09 10:59:36.956 (UTC+0100)] info/session	Connection 5 | SecureChannel 3 | Session 5807e0e5-8f9a-54bf-a2fa-dfbd8db8bb3b | ActivateSession: Session activated

Worked flawlessly, do you have further information ?

thedraft, 2018-03-09:
The client log says
```
9 Mar 10:08:22 - [error] [OPCUA-IIoT-Browser:8276349b.8694c8] Error:  ServiceResult is BadIdentityTokenInvalid (0x80200000)
9 Mar 10:08:28 - [error] [OPCUA-IIoT-Browser:8276349b.8694c8] Error: Session Not Ready To Browse
```

jpfr, 2018-03-09:
UAExpert works flawless.
A client based on node-opcua did not connect.
https://github.com/node-opcua/opcua-commander

The client returns no logs. Also in verbose mode.
The server returns
```
[2018-03-09 11:34:32.196 (UTC+0100)] info/channel       Connection 5 | SecureChannel 3 | Opened SecureChannel
[2018-03-09 11:34:32.271 (UTC+0100)] info/session       Connection 5 | SecureChannel 3 | Session 2643cd78-831b-8cb7-1221-d25d909919d1 | ActivateSession: Could not generate a server nonce
[2018-03-09 11:34:32.333 (UTC+0100)] info/network       Connection 5 | Closed by the server
[2018-03-09 11:34:40.830 (UTC+0100)] info/channel       Connection 0 | SecureChannel 3 | SecureChannel has timed out
[2018-03-09 11:34:50.829 (UTC+0100)] info/session       Connection 0 | SecureChannel 0 | Session 23dcd6aa-34f8-017a-a22a-1231c0ccbc6b | Session has timed out
```

jpfr, 2018-03-09:
By the way, I get the same result when connecting without encryption.

jpfr, 2018-03-09:
I tried with the Softing Client.
Anoymous works with encryption.
Username/Password gives `ActivateSession: Could not generate a server nonce`.

@Infinity95 
Any idea why the nonce is connected to the UserLoginPolicy?

This is now a blocker for 0.3.
We should lock this down with many many unit tests.

jpfr, 2018-03-09:
The same with UAExpert.
`Could not generate a server nonce` with Username/Password.

Infinity95, 2018-03-09:
It has nothing to do with nonces.
Currently this line is the blocker:
```
/* TODO: Support encrypted username/password over unencrypted SecureChannels */
        if(userToken->encryptionAlgorithm.length > 0)
            return UA_STATUSCODE_BADIDENTITYTOKENINVALID;
```
(Line 66 in accesscontrol_default.c)

This simply isn't implemented yet. Refer to Spec Part 4 v1.03 pages 148 onwards.

Since by default the SecurityPolicyUri is set to NULL in the UserTokenPolicies,
[image]
The used encryption Algorithm is the one of the SecureChannel, which cannot be decrypted at the moment, since it is not implemented yet.

[image]
See page 150 part 4 v1.03 as well.

bobbthebuilder, 2018-03-09:
encryption and authentification with username/password (not anonymous) is not by default supported right now and results in "BadIdentityTokenInvalid". What you currently need to do is to write your own activateSession_custom method and let your server config use them (i.e. setting config->accessControl.activateSession = activateSession_custom. I mentioned some details about that in #1634

jpfr, 2018-03-09:
> /* TODO: Support encrypted username/password over unencrypted SecureChannels */
        if(userToken->encryptionAlgorithm.length > 0)
            return UA_STATUSCODE_BADIDENTITYTOKENINVALID;

Thanks for digging this up. Can we add better logging?
Users should not have to dig up this issue to see that
- it is not implemented
- we are working on it

Infinity95, 2018-03-09:
We can set the default UserTokenPolics to SecurityPolicy#None. That shouldn't require too many changes and enable user/password authentication on SecureChannels with policies other than #None.

jpfr, 2018-03-10:
Fixed via  #1646 and

closed 2018-03-10 by jpfr