open62541/open62541 #1661: Maximum recursion depth

Infinity95, 2018-03-15:
Currently the maximum recursion depth is set as the following:
```
#define UA_ENCODING_MAX_RECURSION 20
```
The Standard says in Part6 §5.1.5
> Decoders shall support at least 100 nesting levels

Was there a reason for setting it at 20?
We should probably increase this to 100 to conform to the standard.

Furthermore there are multiple occasions where a `BADENCODINGERROR` is returned when it should actually be a `BADDECODINGERROR`

Pro, 2018-03-16:
Thanks for the hint. When we added the limit, we didn't know that there is a specific limit mentioned in the spec, so this value should be 100.
Regarding the `BADENCODINGERROR`: can you pinpoint the exact locations (or submit a PR)?

jpfr, 2018-03-16:
I'm not sure how much memory this takes on the stack.
A few hundred for each recursion seems realistic.
This could be problematic for memory-constrained embedded systems.

We could make the recursion depth an advanced configuration option.

jpfr, 2018-09-01:
Fixed in #2035.

closed 2018-09-01 by jpfr