open62541/open62541 #1672: Secure channel doesn't get renewed automatically after expiry of its lifetime

SenthilKumarGS, 2018-03-19:
--------------------------------------------------------------------------------

Description
===========
After connection establishment and a subsequent 10 mins of inactivity, when client performs browse /read/any operation, it receives 'UA_CLIENTSTATE_DISCONNECTED' state through state callback.

Note: UA_ClientConfig.secureChannelLifeTime is set to 10mins.
Client application doesn't not manually renew the secure channel.

Background Information / Reproduction Steps
===========================================
1. Client and server sample running on the same machine.
2. Start the server.
3. Start the client and establish connection with the server.
4. Keep the connection inactive for little more than 10mins.
5. Perform any operation from client (read/browse). I performed browse.
Internally, UA_Client_manuallyRenewSecureChannel() gets called.
6. Client receives state callback with state 'UA_CLIENTSTATE_DISCONNECTED'.

Note: UA_Client_manuallyRenewSecureChannel() gets called after the lifetime configured in 'UA_ClientConfig.secureChannelLifeTime'.
I thought this would renew the secure channel internally as the lifetime has expired.
Should client application explicitly call "UA_Client_manuallyRenewSecureChannel" before the lifetime (10mins)?

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Master branch (2be85c43813f7bb0c8fd420b93ab469427ccb4cf) Amalgamated version
 - [ ] Other OPC UA SDKs used (client or server): No
 - [ ] Operating system: Ubuntu 64-bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-03-19:
As far as I understood it (I'm not that deep into the spec in that part) the secure channel has a lifetime of 10mins. After that time it is automatically closed to free up (unused) resources.

If there is a new interaction after this timeout the client should automatically renew the secure channel. It looks like there is a reconnect missing somewhere.
@StalderT  could you have a look at it since you edited that code recently and probably know it quite well?

StalderT, 2018-03-19:
The behavior is correct. If there is no activity for 10 minutes the connection is closed by the server. If needed I can propose a PR for 0.3 that automatically renew the channel in a background task (here : https://github.com/open62541/open62541/blob/fdfb3e2a26651eb433c7a0360d7f7ab796d8f6b0/src/client/ua_client.c#L426)

for 0.4 @jpfr want to do that with https://github.com/open62541/open62541/pull/1508

StalderT, 2018-03-19:
https://github.com/open62541/open62541/pull/1598 can also be used to prevent the issue

closed 2018-03-19 by jpfr

SenthilKumarGS, 2018-03-20:
Thanks a lot for such a timely response.
@StalderT , I am trying to understand how #1598 can be used. Could you please help me know how does it work? Does client application require calling 'UA_Client_runAsync' periodically to renew the secure channel?

StalderT, 2018-03-21:
> @StalderT , I am trying to understand how #1598 can be used. Could you please help me know how does it work? Does client application require calling 'UA_Client_runAsync' periodically to renew the secure channel?

Yes ```UA_Client_runAsync```, (like ```UA_Server_run``` for server part) is the main loop and must be call often as possible.