open62541/open62541 #1675: Suggestion: adding an additional WriteMask to UA_Node

aDogCalledSpot, 2018-03-19:
Description
=========
Adding an additional WriteMask to `UA_Node` in order to be able to stop the server itself from changing things it shouldn't. 

Background Information
====================
Currently events are being implemented as nodes in the server. The developer has the possibility to "create" an event, then manually set certain attributes (as you would with an ObjectNode) before "triggering" the event.

Before triggering the event, it should not be possible to manually set attributes such as, for example, the EventId. After triggering the event it should not be possible to alter anything.

The current WriteMask on a node will not suffice for this use case because using `adminSession` would 
simply override it and the data could be changed anyway. Adding a second mask and making sure that isn't blocking `adminSession` every time it is checked whether the current session is `adminSession` could resolve this problem.

Pro, 2018-03-19:
I discussed this with @aDogCalledSpot. For the events implementation the data of the event is stored in the nodeset and then used to generate the event data. We would need a way to block any (un-)intentional changes to this event object node.
Adding an additional serverAccessLevel to `UA_Node` could solve this issue. It allows to restrict access (delete, edit) even for the `adminSession` on the node.

@open62541/open62541-core any comments against this implementation?

jpfr, 2018-03-19:
> Currently events are being implemented as nodes in the server. 

Wait, what? Events are nodes? I thought that events were emitted by nodes.
Sorry for being a noob on the subject.
Does the event payload have attributes like a node or ist it just a list of variants (with some more metadata)?

> the data of the event is stored in the nodeset and then used to generate the event data

I parse this as a circular definition. Can you unpack?

> @open62541/open62541-core any comments against this implementation?

There are always pros and cons. 😄

In the past, it worked quite well to make the C API first and then add the representation in the information model. Something like the model-view-controller pattern.

Pro, 2018-03-19:
> Wait, what? Events are nodes? I thought that events were emitted by nodes.
Sorry for being a noob on the subject.

No problem, I also took us some time to wrap our head around all that events stuff, unfortunately the spec is not so detailed here. So I will try to explain it, don't hesitate to ask more questions!

There are various factors due to which we (@aDogCalledSpot and myself) came to the conclusion that events are basically instances of event types:

1. A plain server executable with `UA_ENABLE_FULL_NS0=ON` gives the following nodes:
[image]
As you can see, the event types are similar structures as the object types. First it was unclear, why it is modeled like this, but then it came clearer: The `BaseEventType` contains all the necessary information an event has to have at least. I.e. `EventId`, `LocalTime`, `Message`, `Severity`, `SourceNode` and a few more. By simply instantiating an ObjectNode of a SubType of the `BaseEventType`, we have a structure where we can store the required data.
This automatically has the advantage, that for specific event types (like the `TransitionEventType`), additional fields are available in the "structure" and can be filled accordingly

2. Events support filters by using select and where clauses (somewhat similar to the SQL selecting columns and defining where conditions). Have a look at the example here https://github.com/open62541/open62541/blob/master/examples/tutorial_client_events.c#L75. It tells the server, that the client is only interested in the `Message` and `Severity` field, and thus the server should only send these. The fields are identified by the BrowseName. Additionally the client can select if it is interested in the value or some other attribute of that field. Therefore we need a concept on the server side to match this browse name to the corresponding event field. Using the object node instance of an event type, we can simply search through the browse names of all the children (properties) and return the corresponding attribute, e.g., its value.

3. By using this concept, a developer can easily define his own event type, by adding subtypes to the `BaseEventType` and then initialize the instance accordingly. The client can then browse any property of that subtype.

So the general concept is the following:

When a developer/user wants to trigger an Event `TransitionEventType` on the Node `Server` the following steps are necessary and will be implemented:

1. Use the method `UA_Server_createEvent(..., TransitionEventType_ID, ...) ` which simply creates an instance of the given event type, including all its child properties, and then returns the newly created node id `myEventId`. Note that this instance will not be shown in UaExpert since it is not referenced via the typical `HasComponent` or `HasProperty`. So it can be seen as a hidden node.
2. With the `myEventId` and the typical node operation services (setAttribute, ...) the user can then set all the desired properties of the event, e.g., also fill his custom fields or add additional ones.
3. At the end, when all the event properties are set, a method `UA_Server_triggerEvent` needs to be called which adds the node id `myEventId` to the corresponding queues in the node and the monitored item, and thus making it available to be read by the client.
4. After the `UA_Server_triggerEvent` is called on `myEventId` that instance of a specific event type must not be changed anymore, to make sure that client querying the event always get the same data and the developer cannot change the event on accident or on purpose.
5. When the event queue is full (overflow) or the source node is deleted, the corresponding (hidden) event nodes will also be deleted.

In our opinion this is the cleanest way to reuse as much as possible and at the same time provide full events functionality.
We first thought about simply using a key/value list for an event, but this does not cover all the necessary functions, i.e., it does not allow the client to select different attributes of a key, but only its value.

>> the data of the event is stored in the nodeset and then used to generate the event data

>I parse this as a circular definition. Can you unpack?

Based on my previous explanation this sentence can also be written as:
`UA_Server_createEvent` is used to create an instance of an event type and that instance (including all its child properties) is stored in the nodeset. When a client sends a publish request, the server iterates over the event queue of the monitored item (i.e. all the node ids) and checks if the event properties match the event filter. If the filter matches, the selected properties are extracted and the event data (which is sent back to the client) is built.

@jpfr we can also have another short telco, where we discuss that concept in a bit more detail.

jpfr, 2018-03-19:
> @jpfr we can also have another short telco, where we discuss that concept in a bit more detail.

I'm a bit short on time. And your ideas seem already quite mature.
Only if you really want some external input or bounce off a few ideas.

> When a client sends a publish request, the server iterates over the event queue of the monitored item (i.e. all the node ids) and checks if the event properties match the event filter. If the filter [...]

So we copy all events to the queue of interested MonitoredItems and filter only later?
That sounds like a lot of extra copying. Can't we apply the filter right away?

If however we don't copy every event into the MonitoredItems, then the event must be stored in the emitting node so the MonitoredItems can pull it from their callback. But since the publish interval can be anything, how do we know when the event can be finally removed?

Pro, 2018-03-20:
> I'm a bit short on time. And your ideas seem already quite mature.
> Only if you really want some external input or bounce off a few ideas.

For now we can develop a first demo and then discuss based on that, let's see.

> So we copy all events to the queue of interested MonitoredItems and filter only later?
> That sounds like a lot of extra copying. Can't we apply the filter right away?

Our Idea was to simply link an event from the emitting node into the monitored item, i.e., the monitored item only contains a list of event node ids. When a publish request is received, the server goes through that list end extracts the filtered data from the referenced event nodes. This allows us to simply remove events from monitored items when they are published, but still keep the data for other monitored items. And additionally for up-propagation we can add the event node id to all the monitored items on the parent nodes and still keep only one instance of the event.

> If however we don't copy every event into the MonitoredItems, then the event must be stored in the emitting node so the MonitoredItems can find it from their callback. But since the publish interval can be anything, how do we know when the event can be finally removed?

The event can be removed in two cases:
- The emitting node is deleted
- All the monitored items on a node (or its parents) are removed. For this case we have to check again in the spec, if it should be possible to receive events from the time, when the monitored item did not exist yet.

aDogCalledSpot, 2018-03-20:
To get back to the original question; are there any suggestions on how sophisticated the WriteMask should be? For the purposes of an event it would suffice to simply deny all write access for the entire node but we could also set write access for more specific things (for example by just using the same WriteMask rules as for the userWriteMask).

aDogCalledSpot, 2018-03-20:
> So in any long-running session with a monitoreditem, we accumulate events indefinitely.

A maximum amount of events in a node can be set in the server config. If an event is removed from a node it is also removed from all it's monitored items.

jpfr, 2018-03-20:
> If a node is supposed to able to generate events without monitored items this wouldn't work.

If nobody is listening, why would we keep the data?

Access to historical events is part of Historical Data Access. Is HDA in the scope of this development?
HDA could be implemented by a "virtual" MonitoredItem that puts all notifications into a database.

Pro, 2018-03-21:
Closing this after discussion via telco.

To summarize:
After triggering an event it's values (variants) will be transferred into all the corresponding monitored items and then the event node will be deleted.

closed 2018-03-21 by Pro