open62541/open62541 #1678: Server closes the connection randomly during concurrent publish request and method call.

SenthilKumarGS, 2018-03-22:
Description
===========
Client subscribes and manually sends publish request once every 10ms. In parallel, it performs method call. Server closes the connection. It's a random issue because server doesn't close the connection for every call request.

Background Information / Reproduction Steps
===========================================
1. Run server and client samples on the same machine.
2. Start server and start client.
3. Client establishes connection and creates a subscription request.
4. Client starts a thread and manually sends publish request once every 10ms.
5. At the same time, client performs a method call.
6. Server closes the connection.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Master branch (2be85c4) Amalgamated version
 - [ ] Other OPC UA SDKs used (client or server): No
 - [ ] Operating system: Ubuntu 64-bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

SenthilKumarGS, 2018-03-22:
**Server Log:-**

[2018-03-22 14:35:40.304 (UTC+0550)] trace/network      Connection 5 | Activity on the socket
[2018-03-22 14:35:40.304 (UTC+0550)] trace/network      Connection 5 | Received a packet.
[2018-03-22 14:35:40.304 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 2 | Process a MSG
[2018-03-22 14:35:40.304 (UTC+0550)] debug/session      Connection 5 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Processing CallRequest

[sqrt(16.00) method called] :: 4.00
[2018-03-22 14:35:40.304 (UTC+0550)] trace/network      Connection 5 | Activity on the socket
[2018-03-22 14:35:40.304 (UTC+0550)] trace/network      Connection 5 | Received a packet.
[2018-03-22 14:35:40.304 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 2 | Process a MSG
[2018-03-22 14:35:40.304 (UTC+0550)] debug/session      Connection 5 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Processing PublishRequest
[2018-03-22 14:35:40.304 (UTC+0550)] debug/session      Connection 5 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Queued a publication message
[2018-03-22 14:35:40.304 (UTC+0550)] debug/session      Connection 5 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | Response on a late subscription
[2018-03-22 14:35:40.304 (UTC+0550)] debug/session      Connection 5 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | Publish Callback
[2018-03-22 14:35:40.305 (UTC+0550)] debug/session      Connection 5 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | Sending a KeepAlive
[2018-03-22 14:35:40.305 (UTC+0550)] debug/session      Connection 5 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | Sending out a publish response with 0 notifications
[2018-03-22 14:35:40.305 (UTC+0550)] trace/network      Connection 5 | Activity on the socket
[2018-03-22 14:35:40.305 (UTC+0550)] trace/network      Connection 5 | Received a packet.
[2018-03-22 14:35:40.305 (UTC+0550)] trace/channel      Connection 5 | SecureChannel 2 | Process a MSG
[2018-03-22 14:35:40.305 (UTC+0550)] debug/session      Connection 5 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Processing CallRequest

[incrementInt32Array({10 20 30 40 50}, 5) method called] :: 15 25 35 45 55
[2018-03-22 14:35:40.305 (UTC+0550)] trace/network      Connection 5 | Activity on the socket
[2018-03-22 14:35:40.305 (UTC+0550)] info/network       Connection 5 | Closed by the server
[2018-03-22 14:35:40.309 (UTC+0550)] debug/session      Connection 0 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | Publish Callback
[2018-03-22 14:35:40.309 (UTC+0550)] debug/session      Connection 0 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | The publish queue is empty
[2018-03-22 14:35:40.309 (UTC+0550)] debug/session      Connection 0 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | Sending a KeepAlive
[2018-03-22 14:35:40.309 (UTC+0550)] debug/session      Connection 0 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | Want to send a publish response;but not possible
[2018-03-22 14:35:40.318 (UTC+0550)] debug/session      Connection 0 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | Publish Callback
[2018-03-22 14:35:40.318 (UTC+0550)] debug/session      Connection 0 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | The publish queue is empty
[2018-03-22 14:35:40.318 (UTC+0550)] debug/session      Connection 0 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | Sending a KeepAlive
[2018-03-22 14:35:40.318 (UTC+0550)] debug/session      Connection 0 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | Want to send a publish response;but not possible
[2018-03-22 14:35:40.328 (UTC+0550)] debug/session      Connection 0 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | Publish Callback
[2018-03-22 14:35:40.328 (UTC+0550)] debug/session      Connection 0 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | The publish queue is empty
[2018-03-22 14:35:40.329 (UTC+0550)] debug/session      Connection 0 | SecureChannel 2 | Session 152ca78d-6003-027c-f3bf-bb7beefeefbe | Subscription 1 | Sending a KeepAlive

**Client Log:-**

[2018-03-22 14:35:40.183 (UTC+0550)] debug/client       Received a response of type 829
[2018-03-22 14:35:40.193 (UTC+0550)] debug/client       Sending a request of type 824
[2018-03-22 14:35:40.194 (UTC+0550)] debug/client       Received a response of type 829
[2018-03-22 14:35:40.204 (UTC+0550)] debug/client       Sending a request of type 824
[2018-03-22 14:35:40.205 (UTC+0550)] debug/client       Received a response of type 829
[2018-03-22 14:35:40.215 (UTC+0550)] debug/client       Sending a request of type 824
[2018-03-22 14:35:40.216 (UTC+0550)] debug/client       Received a response of type 829
[2018-03-22 14:35:40.226 (UTC+0550)] debug/client       Sending a request of type 824
[2018-03-22 14:35:40.227 (UTC+0550)] debug/client       Received a response of type 829
[2018-03-22 14:35:40.237 (UTC+0550)] debug/client       Sending a request of type 824
[... cut ...]
------------------------------------------------------
[2018-03-22 14:35:40.303 (UTC+0550)] debug/client       Sending a request of type 710
[2018-03-22 14:35:40.303 (UTC+0550)] debug/client       Sending a request of type 824
[2018-03-22 14:35:40.304 (UTC+0550)] debug/client       Received a response of type 715
[2018-03-22 14:35:40.304 (UTC+0550)] debug/client       Sending a request of type 710
[2018-03-22 14:35:40.305 (UTC+0550)] info/client        Error receiving the response with status code BadRequestHeaderInvalid

SenthilKumarGS, 2018-03-23:
Client thread calls 'UA_Client_Subscriptions_manuallySendPublishRequest' to send publish request once every 10ms. In between, main thread of client calls 'UA_Client_call' to trigger method call.
This is when the connection gets dropped randomly.

Unfortunately i can't share code from my company network. Sorry.
But i will try to write a sample and share it along with the network dump from my personal network.

FYI, i am not using the latest master branch code.
The one which i am using is an amalgamated version of a couple of months old snapshot.
Commit hash: 2be85c43813f7bb0c8fd420b93ab469427ccb4cf
I will try with the latest code and share the results here.

UBehrens, 2020-07-22:
I'm facing similar (and maybe related) problems with an old (0.3.0) version of the library. However there is a kind of unusual setup:

1: Build a Connection between Server and Client
2: Set up two(!) Subscriptions between the Server and the Client, both having an inactivity and delete callback installed.
3: Use the Connection every 5s to send Publish requests from Client to Server and receive Responses.
4: After arbitrary time, I receive the 'BadRequestHeaderInvalid' message, maybe from a Subscription?
5: Both Subscriptions from Step 2 are being deleted (sounds reasonable, since they have not been used for a while).

6: The client is being disconnected (UA_CLIENTSTATE_DISCONNECTED being sent).

Why does this happen? A closed Subscription should not be the same as a closed Connection, IMHO...? Is a Connection closed ASA all Subscriptions are being deleted, regardless of the Connection being used in other contexts like Publish/Response? I have used UA_Client_runAsync() in the main loop to ensure that the Response events are read regularly from the Server. Did I miss something like telling the Server not to close the Connection on the first possible opportunity?

On the other hand, if I call 'UA_Client_manuallyRenewSecureChannel()' this returns UA_STATUSCODE_GOOD and I interpret this as the connection having been re-instantiated successfully. But why would this be necessary at all at this point?

Finally, I run the Publish/Response loop in an independent thread, but I don't think that's the major issue here. What really confuses me is how to distinguish a closing Connection from a closing Subscription and the related consequences.

Thanks a lot in advance,

Uwe

teccheck, 2025-04-16:
v0.3 is quite old at this point, there have been a lot of improvements and the CI covers several scenarios. As there are no recent comments, I assume this is resolved

closed 2025-04-16 by teccheck