open62541/open62541 #1685: Possible DOS attack?

Infinity95, 2018-03-23:
Description
===========
I believe it is currently possible to create a DOS attack by creating lots of SecureChannels with a session in them.
The sequence is as follows:
HEL -> ACK -> OPN Request -> OPN Response -> CreateSessionRequest -> repeat

After 40 times this will fill up the maximum number of SecureChannels.
This is only possible if the Server doesn't use a trust list or if the attacker is trusted.

Any thoughts if this is a relevant issue and how to solve it (except from using a trust list of course)?
Might be a problem if a client is broken or something?

jpfr, 2018-03-23:
Imho we can have a shorter timeout for sessions that have not been activated.
Or we remove an inactivated session if there are too many...

But we probably cannot circumvent a client keeping many sessions open...

Stasik0, 2018-03-23:
Is there something in the standard? Like max session per client?

Pro, 2018-11-06:
Duplicate of #1808 ?