open62541/open62541 #1698: Compiling error on 0.3 and large binary

eliasnaslund, 2018-03-29:
Description
===========
I'm compiling tutorial_server_datasource.c exactly as it is with the single source distribution and get error when compiling with all of the features disabled.

If I manually alter the code to make it compile the binary is over 100K larger then the same example with v0.2. Is it possible to decrease the overhead? both constant data and code size have increased a lot even though no options is enabled.

Background Information / Reproduction Steps
===========================================
```
/**
 * Feature Options
 * ---------------
 * Changing the feature options has no effect on a pre-compiled library. */
#define UA_LOGLEVEL 800
//#define UA_ENABLE_METHODCALLS
//#define UA_ENABLE_NODEMANAGEMENT
//#define UA_ENABLE_SUBSCRIPTIONS
/* #undef UA_ENABLE_MULTITHREADING */
/* #undef UA_ENABLE_ENCRYPTION */

/* Advanced Options */
//#define UA_ENABLE_STATUSCODE_DESCRIPTIONS
//#define UA_ENABLE_TYPENAMES
/* #undef UA_ENABLE_DETERMINISTIC_RNG */
/* #undef UA_ENABLE_GENERATE_NAMESPACE0 */
/* #undef UA_ENABLE_NONSTANDARD_UDP */
//#define UA_ENABLE_DISCOVERY
/* #undef UA_ENABLE_DISCOVERY_MULTICAST */
//#define UA_ENABLE_DISCOVERY_SEMAPHORE
/* #undef UA_ENABLE_UNIT_TEST_FAILURE_HOOKS */
/* #undef UA_ENABLE_VALGRIND_INTERACTIVE */
//#define UA_VALGRIND_INTERACTIVE_INTERVAL 1000

/* Options for Debugging */
/* #undef UA_DEBUG */
/* #undef UA_DEBUG_DUMP_PKGS */
```

I have tried both v0.3rc1 and 0.3-rc1-62-ge2dde65. With both versions I get:
```
gcc -std=c99 -Os tutorial_server_datasource.c open62541.c -o tutorial_server_datasource -s
open62541.c:41065:5: warning: initialization makes integer from pointer without a cast [-Wint-conversion]
     NULL,  /*.clientContext */
     ^
open62541.c:41065:5: note: (near initialization for ‘UA_ClientConfig_default.outStandingPublishRequests’)
open62541.c:41067:5: warning: excess elements in struct initializer
     10 /* .outStandingPublishRequests */
     ^
open62541.c:41067:5: note: (near initialization for ‘UA_ClientConfig_default’)
/tmp/ccDNkP15.o: In function `ua_namespace0':
open62541.c:(.text+0x255fd): undefined reference to `UA_Server_addMethodNode_finish'
collect2: error: ld returned 1 exit status
```

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): v0.3rc1 & 0.3-rc1-62-ge2dde65
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 16.04 32 bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

StalderT, 2018-03-29:
For compilation issues :

First issue fixed in https://github.com/open62541/open62541/pull/1699

The second is that the nodeset_compiler generates a method while UA_ENABLE_METHODCALLS is set to OFF : 

```
/* GetMonitoredItems - ns=0;i=11492 */

static UA_StatusCode function_ua_namespace0_12_begin(UA_Server *server, UA_UInt16* ns) {

UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_MethodAttributes attr = UA_MethodAttributes_default;
attr.executable = true;
attr.userExecutable = true;
attr.displayName = UA_LOCALIZEDTEXT("", "GetMonitoredItems");
attr.description = UA_LOCALIZEDTEXT("", "");
attr.writeMask = 0;
attr.userWriteMask = 0;
retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_METHOD,
UA_NODEID_NUMERIC(ns[0], 11492),
UA_NODEID_NUMERIC(ns[0], 0),
UA_NODEID_NUMERIC(ns[0], 0),
UA_QUALIFIEDNAME(ns[0], "GetMonitoredItems"),
UA_NODEID_NULL,
(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_METHODATTRIBUTES],NULL, NULL);
retVal |= UA_Server_addReference(server, UA_NODEID_NUMERIC(ns[0], 11492), UA_NODEID_NUMERIC(ns[0], 47), UA_EXPANDEDNODEID_NUMERIC(ns[0], 2253), false);
return retVal;
}

static UA_StatusCode function_ua_namespace0_12_finish(UA_Server *server, UA_UInt16* ns) {

return UA_Server_addMethodNode_finish(server,
UA_NODEID_NUMERIC(ns[0], 11492)
, NULL, 0, NULL, 0, NULL);
}
```

Pro, 2018-03-29:
@StalderT thanks again!

In the nodeset compiler we know which method we are currently printing, i.e. `UA_NODECLASS_METHOD`. We can simply wrap the printing of these nodes into a '#ifdef UA_ENABLE_METHODCALLS`

@eliasnaslund are you using the FULL NS0 or the minimal NS0? Can you post how you called CMake. There are also some additional tweaks which can be performed to reduce the size, i.e. disalbe type names or decreasing the loglevel. We definitely need to investigate and document how one can create a minimal size binary.

eliasnaslund, 2018-03-29:
@Pro I'm not calling CMake at all but just using the open62541.c/h files already generated. I can definitely generate my own if that would reduce the size. The loglevel, typenames and statuscode description is disabled in open62541.h. See Background Information in my original post. I would love a document how one can create a minimal size binary.

Pro, 2018-03-29:
I created a new issue for the size doc #1700

closed 2025-04-16 by teccheck