open62541/open62541 #1700: Write doc for how to create a minimal binary and check how to minimize binary file

Pro, 2018-03-29:
There's no documentation on how to create a minimum size binary.

We should add a document on that, especially for resource limited devices.
It cloud include notes about:
- Use / customize the minimal nodeset NS0
- Decrease loglevel
- Disable typenames
- Set compiler optimization options (e.g. `-Os`)

Additionally we should analyze the generated binary and see where most of the size is used up (e.g., static objects, nodeset compiler output, ...)

More ideas are welcome!

jpfr, 2018-03-29:
Even with all the optimizations, the binary size more than doubled since we added nodeset definitions that are generated from xml.

It was ~80kb. Now ~200kb.

Pro, 2018-03-29:
We should try to minimize our minimal nodeset XML and remove some (unnecessary?) nodes. When doing that, one should make sure that the file is NOT reordered, to be able to see the removed items in a diff, i.e. lines should only be deleted, but not moved.

jpfr, 2018-03-29:
#1697 is adding nodes to the minimal nodeset. ☹️ 
They are required to pass the CTT.

👍 for minimizing the minimal xml
We can also add a "nano" option that does not pass the CTT but makes most clients happy.

jpfr, 2018-03-29:
By the way, there are docs for minimizing the binary size.
https://open62541.org/doc/current/building.html#minimizing-the-binary-size

In place since a year or so...

jpfr, 2018-03-29:
See #1638 for a way to further reduce the binary size...

Stasik0, 2018-03-29:
@jpfr do you try nano or 2017 nano? As I told yesterday. It it a ctt bug?

The support of Diagnostic Objects and Variables is optional for this Profile despite it being defined as “mandatory” in UA Part 5. Support of Diagnostic Objects and Variables is mandatory in some higher level Profiles. Exposing types in the AddressSpace is optional for this Profile except if custom types (i.e. types that are derived from well-known ObjectTypes, VariableTypes, ReferenceType or DataTypes) are used. Exposing all supported types in the AddressSpace is mandatory in some higher level Profiles. This profile supersedes the "Nano Embedded Device Server Profile".
 URI	http://opcfoundation.org/UA-Profile/Server/NanoEmbeddedDevice2017

Stasik0, 2018-03-29:
Can we ommit more strings in the nodeset? Like not setting locale, or using the same locale string or ommitng Browsename/displayname for ns0? Will ctt eat that?

jpfr, 2018-03-30:
Hey @Stasik0 ,

thanks for the heads up!
The latest CTT download is for 1.03 and does not have the 2017 profiles included.
So we cannot easily test.

It seems they want to enable more lightweight servers.
But actually, the 2017 Core Server Facet requires a number of additional feature sets:

- Address Space Atomicity (mandatory)
- Address Space Full Array Only (mandatory)
- Base Info Estimated Return Time (optional)
- Base Info Selection List (optional)

I think we may require some time implementing those.
First, they rely on additional node attributes from 1.04 (AccessLevelEx).
Second, most of the "type checking" we perform internally assumes the existence of the type definitions.

eliasnaslund, 2018-04-03:
@Stasik0 Actually, if compiling with gcc -Os, the compiler is smart enough to use the same locale for all strings. en_US is found only once in .rodata when examining with objdump -s.

Pro, 2018-04-26:
Also related: #101

Pro, 2018-07-11:
PR #1742 also added the CMake option `UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS` to disable the generation of node description from .xml

closed 2021-09-23 by jpfr