open62541/open62541 #1702: client_secure_new problem

etm, 2018-03-29:
0) https://github.com/open62541/open62541/commit/da64add5c0d5eb0fda2ea7f4f5e04cc20e237ab9
1) We use an unmodified examples/encryption/client_basic256sha256.c
2) We utilize DER acertificate and a DER key that correctly work in an freeopcua client
3) We use no trusted list

The problem occurs in file src/client/ua_client.c method UA_Client_secure_init:

retval = client->securityPolicy.channelModule.newContext(securityPolicy, remoteCertificate,
                                                             &client->channel.channelContext);

retval has the value of 0x80180000 (UA_STATUSCODE_BADCERTIFICATEUSENOTALLOWED).

Any hints what the problem could be?

StalderT, 2018-03-30:
Do you use server_basic256sha256 as server ?

Can you provide your DER certificate and your DER key for client and server?

etm, 2018-03-30:
opcua server runs on an sinumerik 840d sl. It is reachable under opc-tcp://tango.wst.univie.ac.at:9200.

the certificate can be found:
http://tango.wst.univie.ac.at/html/opc2mqtt.der
the key can be found:
http://tango.wst.univie.ac.at/html/opc2mqtt_key.der

Username and password are standard:
UA_Client_connect_username(client, "opc.tcp://tango.wst.univie.ac.at:9200","OpcUaClient","SUNRISE");

StalderT, 2018-03-30:
Just to make a test. Can you try to change here : 

https://github.com/open62541/open62541/blob/master/plugins/ua_securitypolicy_basic256sha256.c#L39

```
#define UA_SECURITYPOLICY_BASIC256SHA256_MINASYMKEYLENGTH 256
```

to 

```
#define UA_SECURITYPOLICY_BASIC256SHA256_MINASYMKEYLENGTH 128
```

etm, 2018-03-30:
Solves the initial issue, but connect_username now returns

UA_STATUSCODE_BADTCPMESSAGETYPEINVALID 0x807e0000 // The type of the message specified in the header invalid.

StalderT, 2018-03-30:
Does it work with UA_Client_connect?

etm, 2018-03-30:
Yes, but when i read a node UA_STATUSCODE_BADNOTREADABLE 0x803a0000 // The access level does not allow reading or subscribing to the Node.

p.s. reading 
UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME);
works without any security. But e.g. UA_NODEID_STRING(2, "/Channel/State/progStatus") does not.

StalderT, 2018-03-31:
The initial issue will be fixed.

Can you open a new issue for connect_username issue. Maybe something like https://github.com/open62541/open62541/issues/1644

jpfr, 2019-02-10:
Seems to be resolved.
Please reopen if required.

closed 2019-02-10 by jpfr