open62541/open62541 #1704: connect_username

etm, 2018-03-31:
0. https://github.com/open62541/open62541/commit/da64add5c0d5eb0fda2ea7f4f5e04cc20e237ab9
1. we try to connect to an opcua server on a sinumerik 840d sl. 
2. server is reachable under opc-tcp://tango.wst.univie.ac.at:9200.
3. certificate: http://tango.wst.univie.ac.at/html/opc2mqtt.der
4. key: http://tango.wst.univie.ac.at/html/opc2mqtt_key.der
5. user/pass. UA_Client_connect_username(client, "opc.tcp://tango.wst.univie.ac.at:9200","OpcUaClient","SUNRISE");

Please not that the changes discussed in https://github.com/open62541/open62541/issues/1702 are not yet committed, the proposed change has to be done by hand before this report can be reproduced.

When using examples/encryption/client_basic256sha256.c to connect to the server connect_username returns UA_STATUSCODE_BADTCPMESSAGETYPEINVALID 0x807e0000 // The type of the message specified in the header invalid.

Infinity95, 2018-03-31:
I tried to connect to the uri you mentioned but the connection is refused.
Is the server behind a firewall?
If making the server available is not an option, you could supply a pcap file with the relevant communication and the symmetric nonces, so that you don't have to supply the server private key.

Have you tried using a different key, that conforms to the keylengths the standard specifies?
(that is 2048 - 4096 bits for basic256sha256)

Infinity95, 2018-03-31:
I believe the issue might be that you are using a non-conformant key length with that workaround which in turn gets rejected by the server (as it should).
This produces an ERR message, which produces a BADTCPMESSAGETYPEINVALID error here:
[image]

So changing your key length should fix the problem.

etm, 2018-04-03:
I generated a new cert/key pair:
* http://tango.wst.univie.ac.at/html/cert.der
* http://tango.wst.univie.ac.at/html/cert_key.der

Still using the unmodified examples/encryption/client_basic256sha256.c with  UA_Client_connect_username(client, "opc.tcp://tango.wst.univie.ac.at:9200","OpcUaClient","SUNRISE");

i get a 0x807e0000 for connect_username, same as with the old key.

regarding the server/firewall: i switched it off for easter weekend, it is online again. So you should be able to test it .

p.s. connecting to a unified automation server with the certificate gives a UserTokenPolicy error. connecting with freeopc python lib as a reference point still works with UA server and Sinumeric.

Infinity95, 2018-04-03:
I tried connecting with UAExpert and the server doesn't seem to support SecurityPolicy#Basic256Sha256.
I haven't been able to try with open62541 yet, but will do so later.
You could try connecting with SecurityPolicy#Basic128Rsa15, since the server seems to support that.

etm, 2018-04-03:
[image]

Discovery of UAExpert thinks Basic256 is a thing. Might be that discovery is not working correctly after tunnelling through firewall.

nonetheless, when using SecurityPolicy#Basic128Rsa15 i get 0x80020000 (UserTokenPolicy problem).

Infinity95, 2018-04-03:
Keep in mind that Basic256 is not Basic256Sha256.
Those are two different policies.
I will check what's wrong whith the UserTokenPolicy.

etm, 2018-04-04:
We can leave the sinumeric online for now if that helps.

jpfr, 2019-02-10:
Fixed on master.

closed 2019-02-10 by jpfr