open62541/open62541 #1709: Invalid Code Generation for Variable with Rank 2 (two dimensional array DataType string)

lehned, 2018-04-02:
Description
===========
A variable node with two dimensional array, generated by the  nodeset compiler, results in an "Segmentation fault"  on deleting the node, after error _BadTypeMismatch_ on _WriteValue_ in _UA_Server_addNode_begin_ for the node.

Background Information / Reproduction Steps
===========================================

1. compile model with `python ../tools/nodeset_compiler/nodeset_compiler.py -e ../deps/Schema/Opc.Ua.NodeSet2.xml -x ../module/issue_array2dim.xml ../module/testmodel` required FULL_NS0

```
<?xml version="1.0" encoding="utf-8"?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://test.org/test_arrayrank2/Types.xsd" xmlns:ua="http://unifiedautomation.com/Configuration/NodeSet.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://test.org/test_arrayrank2/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="String">i=12</Alias>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ua:ModelInfo Tool="UaModeler" Hash="l3n1gs1K2HWADqWzc8INWQ==" Version="1.5.2"/>
        </Extension>
    </Extensions>
    <UAObjectType NodeId="ns=1;i=1002" BrowseName="1:TestObject">
        <DisplayName>TestObject</DisplayName>
        <References>
            <Reference ReferenceType="HasComponent">ns=1;i=6001</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
        </References>
    </UAObjectType>
    <UAVariable DataType="String" ParentNodeId="ns=1;i=1002" ValueRank="2" NodeId="ns=1;i=6001" BrowseName="1:Array2dim" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>Array2dim</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1002</Reference>
        </References>
        <Value>
            <uax:ListOfString>
                <uax:String></uax:String>
            </uax:ListOfString>
        </Value>
    </UAVariable>
</UANodeSet>
```
2. create cmake config vi open62541-config.cmake

```
set(UA_ENABLE_AMALGAMATION ON CACHE BOOL "")
set(UA_ENABLE_METHODCALLS ON CACHE BOOL "")
set(UA_ENABLE_GENERATE_NAMESPACE0 ON CACHE BOOL "")
set(CMAKE_BUILD_TYPE Debug CACHE STRING "")
set(UA_ENABLE_FULL_NS0 ON CACHE BOOL "")
```
3.  compile open62541, to generate amalgamation files open62541.c/.h

```
cd open62541
rm -rf CMakeFiles
rm -f CMakeCache.txt
cmake -C ../open62541-config.cmake .
```
4.   compile testprogram test.c
```

#include <stdint.h>
#include "open62541.h"
#include "testmodel.h"

struct st_private_opcua_data
{
  UA_ServerConfig       *config;
  UA_ServerNetworkLayer nl;
  UA_Server             *server;
  volatile UA_Boolean   running;
  uint32_t              ulPort;
};

struct st_private_opcua_data g_stOPCUA;

int main(int argc, char **argv) {

	struct st_private_opcua_data *pPriv = &g_stOPCUA;

    /* Create a server with one network layer listening on port 4840 */
    pPriv->config = UA_ServerConfig_new_minimal((pPriv->ulPort==0)?4840:pPriv->ulPort, NULL);
    //UA_ServerConfig_set_customHostname(pPriv->config, UA_STRING_ALLOC(cURI));
    /** load server, initiate new server instance */
    pPriv->server = UA_Server_new(pPriv->config);

    testmodel(pPriv->server);

    pPriv->running = 1;

    /* Run the server loop */
    switch (UA_Server_run(pPriv->server, &pPriv->running))
    {
      case UA_STATUSCODE_GOOD:
        return 0;
      default:
  	    return 1;
    }

	/* deallocate certificate's memory */
	UA_Server_delete(pPriv->server);
	pPriv->nl.deleteMembers(&pPriv->nl);
}
```

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): v0.2-996-g37e8654d
 - [x] Other OPC UA SDKs used (client or server): server
 - [x] Operating system: mingw
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
`[2018-04-02 17:09:50.676 (UTC+0300)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | WriteRequest returned status code BadTypeMismatch
[2018-04-02 17:09:50.676 (UTC+0300)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Type-checking the variable node failed with error code BadTypeMismatch
[2018-04-02 17:09:50.677 (UTC+0300)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Node could add parent references with error code BadTypeMismatch
`
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

jpfr, 2018-04-03:
How do you expect the result to look like?

```xml
        <Value>
            <uax:ListOfString>
                <uax:String></uax:String>
            </uax:ListOfString>
        </Value>
```

So you want to get a matrix (value rank 2).
The defined value is an array with one element: An empty string.
This could indeed be a 1x1 matrix. But you define no array dimensions.

- Type checking the node fails
- Creating the server fails
- UA_Server_new() returns a null ptr

@Pro @ichrispa Do we have a way to define array dimensions in the XML nodesets?

The example is nice and self-contained. We could keep a list of nodesets in /tests/nodeset_compiler and autogenerate a test executable for each of them with cmake.

Pro, 2018-04-03:
> @Pro @ichrispa Do we have a way to define array dimensions in the XML nodesets?

It should be possible to define array dimensions by specifying the `ValueRank="..."` attribute of a `UAVariable` or `UAVariableType` XML line:
```xml
<UAVariable NodeId="i=12169" BrowseName="EnumValues" ParentNodeId="i=120" DataType="i=7594" ValueRank="1">
```

This should then be used as array dimension:
https://github.com/open62541/open62541/blob/8066c68aa254366dbed2b9cc5e49e89e7fc3ea9c/tools/nodeset_compiler/backend_open62541_nodes.py#L123

I have to investigate that further, but in general it should work (as it is the most favourite saying of a programmer :smile: )

jpfr, 2018-04-03:
The value rank does not specify the array dimensions.

It value rank only defines restrictions for allowed array dimensions.
The array dimensions can further define the size of each dimension. With 0 as a wild card.

lehned, 2018-04-05:
Hi,
thanks for the reply. I know there is a mistake in this example model, the modellers are buggy also.

How do you expect the result to look like?
I expected an empty node with empty value applied. I expected to resize the array in runtime by applying values.  
It shouldn't lead to an segmentation fault.

What should a fix do in this case?
A fix should be applied in the nodeset compiler, generating an empty array (depending on dimension) which should sufficient for the copy requirement. Otherwise when no value copy isn't required, only an empty node of this type.

The _ValueRank_ only describes the type of array for the node.  The _ArrayDimension_ are should be used for the memory allocation and value description. In my example invalid therefore, error or default {0,0}.

Best regards!

teccheck, 2025-04-16:
This should be solved now, as we have xml encoding in the stack, and handover the value xml from the nodeset file to the c code