open62541/open62541 #1710: UA client connect to Kepware server error;

mmesa2016, 2018-04-03:
Test OPC UA client test to connect to OPC UA KEPServerEX 5 Server version 5.20. Try to connect using username and password. In the log show "No suitable endpoint found". I' am configure OPC UA endpoint security policies None.
[image]

jpfr, 2018-04-03:
Our Github issue template asks for some background information.
Can you give us more details to reproduce?

Which version of open62541? One of our example servers?
Can you connect with UAExpert?

StalderT, 2018-04-04:
No encryption of username/password is supported at the moment #934

KEPServer accepts only username/password encryption.

[image]

For now, you can enable KEPServer anonymous access : https://forum.inductiveautomation.com/t/faulted-connection-with-kepware/15710/5

closed 2018-04-04 by jpfr

mmesa2016, 2018-04-04:
Description
Test OPC UA client test (open62541 client) to connect to OPC UA KEPServerEX 5 Server version 5.20. Try to connect using username and password. In the log show "No suitable endpoint found". I' am configure OPC UA endpoint security policies None. I' am use UAExpert and connection is successfull. OPC UA KEPServerEX 5 Server enable anonymous access.

 open62541 Version:
 https://github.com/open62541/open62541/tree/master

 Operating system:
 Wwindows 8.1 and build MinGW

 Logs:
1 endpoints found
URL of endpoint 0 is opc.tcp://127.0.0.1:49320
[2018-04-04 10:54:35.505 (UTC-0400)] error/client	No suitable UserTokenPolicy found for the possible endpoints

Code:
 client = UA_Client_new(UA_ClientConfig_default);
 retval = UA_Client_connect_username(client, CONNECTION_STRING, "admin", "admin");

StalderT, 2018-04-04:
> OPC UA KEPServerEX 5 Server enable anonymous access.

So use ```UA_Client_connect``` instead of  ```UA_Client_connect_username```.

mmesa2016, 2018-04-04:
I' am enbled OPC UA KEPServerEX 5 Server anonymous access. For use method UA_Client_getEndpoints in open62541 client, discovery endpoints.