open62541/open62541 #1729: Error compile example server_register.c

marioricardojr, 2018-04-09:
i compile this example using this command:
gcc -std=c99 open62541.c server_register.c -o serveregister

and returns this error: 
server_register.c: In function ‘main’:
server_register.c:98:61: warning: passing argument 2 of ‘UA_Server_addPeriodicServerRegisterCallback’ from incompatible pointer type [-Wincompatible-pointer-types]
         UA_Server_addPeriodicServerRegisterCallback(server, clientRegister, DIS
                                                             ^
In file included from server_register.c:12:0:
open62541.h:11582:1: note: expected ‘const char *’ but argument is of type ‘UA_Client * {aka struct UA_Client *}’
 UA_Server_addPeriodicServerRegisterCallback(UA_Server *server, const char* disc
 ^
server_register.c:14:35: warning: passing argument 3 of ‘UA_Server_addPeriodicServerRegisterCallback’ makes integer from pointer without a cast [-Wint-conversion]
 #define DISCOVERY_SERVER_ENDPOINT "opc.tcp://localhost:4840"
                                   ^
server_register.c:98:77: note: in expansion of macro ‘DISCOVERY_SERVER_ENDPOINT’
 UA_Server_addPeriodicServerRegisterCallback(server, clientRegister, DISCOVERY_S
                                                                     ^
In file included from server_register.c:12:0:
open62541.h:11582:1: note: expected ‘UA_UInt32 {aka unsigned int}’ but argument is of type ‘char *’
 UA_Server_addPeriodicServerRegisterCallback(UA_Server *server, const char* disc
 ^
server_register.c:99:69: warning: passing argument 5 of ‘UA_Server_addPeriodicServerRegisterCallback’ makes pointer from integer without a cast [-Wint-conversion]
                                                     10 * 60 * 1000, 500, NULL);
                                                                     ^
In file included from server_register.c:12:0:
open62541.h:11582:1: note: expected ‘UA_UInt64 * {aka long long unsigned int *}’ but argument is of type ‘int’
 UA_Server_addPeriodicServerRegisterCallback(UA_Server *server, const char* disc
 ^
server_register.c:98:9: error: too many arguments to function ‘UA_Server_addPeriodicServerRegisterCallback’
         UA_Server_addPeriodicServerRegisterCallback(server, clientRegister, DIS
         ^
In file included from server_register.c:12:0:
open62541.h:11582:1: note: declared here
 UA_Server_addPeriodicServerRegisterCallback(UA_Server *server, const char* disc
 ^
server_register.c:126:53: warning: passing argument 2 of ‘UA_Server_unregister_discovery’ from incompatible pointer type [-Wincompatible-pointer-types]
     retval = UA_Server_unregister_discovery(server, clientRegister);
                                                     ^
In file included from server_register.c:12:0:
open62541.h:11559:1: note: expected ‘const char *’ but argument is of type ‘UA_Client * {aka struct UA_Client *}’
 UA_Server_unregister_discovery(UA_Server *server, const char* discoveryServerUr

Solutions? Thanks

marioricardojr, 2018-04-09:
I'm lost in regard to the examples. is there any tutorial for discovery server?
i find this link, but I managed to compile
https://github.com/open62541/open62541/issues/972

Pro, 2018-04-09:
You have to make sure that the examples you are using match the version of your open62541.c file. 

Examples for master:
https://github.com/open62541/open62541/tree/master/examples/discovery

Examples for 0.3:
https://github.com/open62541/open62541/tree/0.3/examples/discovery

marioricardojr, 2018-04-09:
now this error on server multicast 
gcc -std=c99 open62541.c server_multicast.c -o server_multicast
server_multicast.c: In function ‘main’:
server_multicast.c:133:5: warning: implicit declaration of function ‘UA_Server_setServerOnNetworkCallback’ [-Wimplicit-function-declaration]
     UA_Server_setServerOnNetworkCallback(server, serverOnNetworkCallback, NULL)
     ^
/tmp/ccpXSuiU.o: na função `main':
server_multicast.c:(.text+0x896): referência indefinida para `UA_Server_setServerOnNetworkCallback'
collect2: error: ld returned 1 exit status

Pro, 2018-04-09:
You are using the discovery multicast functionality, which is not included in the amalgamated release, since it also depends on the mdns library. To use discovery multicast, you need to build from source, i.e. something like this
```
git clone https://github.com/open62541/open62541.git
cd open62541
git checkout 0.3
git submodule init
git submodule update
mkdir build
cd build
cmake -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=OFF -DUA_BUILD_EXAMPLES=ON ..
make -j
```
This will also build the server_register.c example code.

marioricardojr, 2018-04-12:
client find servers return this
./client_find_servers 
[2018-04-12 14:29:59.689 (UTC-0300)] info/client	Received a ServiceFault response
[2018-04-12 14:29:59.689 (UTC-0300)] error/server	Could not call FindServersOnNetwork service. Is the discovery server started? StatusCode BadServiceUnsupported

Pro, 2018-04-12:
Oh sorry, my previous post was not correct. I had `-DUA_ENABLE_DISCOVERY_MULTICAST=OFF`, instead you should use `-DUA_ENABLE_DISCOVERY_MULTICAST=ON`. I updated the post too