open62541/open62541 #1733: User selectable adding of optional childs from a TypeNode

lehned, 2018-04-12:
Description
===========
Hello, 
some _TypeNodes_ could have references like _HasModellingRule_ as _Optional_. In this case especially they will be skipped at the creation of the instance.
```
ua_service_nodemanagment.c - copyChildNode():
...
/* Is the child mandatory? If not, skip */
    if(!isMandatoryChild(server, session, &rd->nodeId.nodeId))
        return UA_STATUSCODE_GOOD;
```
There are several ways to add _Optional_ reference:
* create them manually, by manual copy form _TypeNode_
* manipulate the  _TypeNode_ after creation from model
* callback for user specific selection.

What should be the proper way?

Best regards!

GeorgeSellner, 2019-03-13:
I am curious regarding this as well.
In the past I have resorted to modifying the model directly in UAModeler and changed to 
`mandatory `

Can anyone point to example code for the 3 mentioned methods?

Pro, 2019-03-14:
It would be nice to have an additional setting for the AddNode method, where one can change the optional to mandatory.

Since types can have a long inheritance hierarchy, the most intuitive way would be to just pass an array of node ids.

The check `if(!isMandatoryChild(server, session, &rd->nodeId.nodeId))` can then compare the node ID and add the optional child if it is in the list.

@open62541/open62541-core any opinions on this?

We would need to add more addNode methods then, e.g. add:

```c
static UA_INLINE UA_StatusCode
UA_Server_addObjectNode_optionals(UA_Server *server, const UA_NodeId requestedNewNodeId,
                        const UA_NodeId parentNodeId,
                        const UA_NodeId referenceTypeId,
                        const UA_QualifiedName browseName,
                        const UA_NodeId typeDefinition,
                        const UA_ObjectAttributes attr,
                        const UA_NodeId *mandatoryChildIds,
                        const size_t mandatoryChildIdsSize,
                        void *nodeContext, UA_NodeId *outNewNodeId) {
// ...
}
```

ja4nm, 2022-10-31:
What is the status with this issue? I see, that there is a callback that user can override, to select which optional properties should be published. Yet, there is no node/user context attribute present in the callback, that way it is not very usbale.

```
server->config.nodeLifecycle.createOptionalChild()

UA_Boolean (*createOptionalChild)(UA_Server *server,
                                      const UA_NodeId *sessionId,
                                      void *sessionContext,
                                      const UA_NodeId *sourceNodeId,
                                      const UA_NodeId *targetParentNodeId,
                                      const UA_NodeId *referenceTypeId);
```

o6-nbeier, 2026-10-20:
Thank you for taking the time to open this issue and describe it so clearly.

The problem has since been fixed on the maintained branches (see 13fa838d54), so we are closing it as resolved.

closed 2026-10-20 by o6-nbeier