open62541/open62541 #1734: Not compiling with Windows (minGW/MSVS)

genericFJS, 2018-04-12:
Description
===========
The minimal example(s) given in the [tutorial/documentation](https://open62541.org/doc/0.3/tutorial_client_firststeps.html) can not be compiled under Windows 7 64bit with gcc (MinGW) or MSVS. The files used can however be compiled with gcc under Ubuntu/Linux (Windows 10 WSL).

Background Information / Reproduction Steps
===========================================
After a few failed attempts to compile, following steps were made to verify there is no error with the (Windows-)installation and reproduce the error:

### MinGW
- Create a fresh install of Windows 7 64bit on a virtual machine (VMware).
- Install MinGW with mingw32-base (includes gcc). Add MinGW to `PATH` environment variable.
- Copy `open62541.c` and `open62541.h` from [open62541 website](https://open62541.org/) into a folder `test`.
- Copy code from [Simple Client Tutorial](https://open62541.org/doc/0.3/tutorial_client_firststeps.html) into the same folder (file `myClient.c`).
- Open command prompt.
- Change into folder `test`
- Run `gcc -std=c99 open6251.c myClient.c -lws2_32 -o myClient.exe`.
- Error :heavy_multiplication_x: 
Error Message:
```
open62541.c: In function 'socket_set_nonblocking':
open62541.c:38482:48: error: 'NO_ERROR' undeclared (first use in this function)
     if(ioctlsocket(sockfd, FIONBIO, &iMode) != NO_ERROR)
                                                ^~~~~~~~
open62541.c:38482:48: note: each undeclared identifier is reported only once for each function it appears in
open62541.c: In function 'socket_set_blocking':
open62541.c:38500:48: error: 'NO_ERROR' undeclared (first use in this function)
     if(ioctlsocket(sockfd, FIONBIO, &iMode) != NO_ERROR)
                                                ^~~~~~~~
open62541.c: In function 'addServerSocket':
open62541.c:38636:42: error: 'IPV6_V6ONLY' undeclared (first use in this function)
        setsockopt(newsock, IPPROTO_IPV6, IPV6_V6ONLY,
                                          ^~~~~~~~~~~
open62541.c: In function 'UA_DateTime_localTimeUtcOffset':
open62541.c:39252:5: warning: implicit declaration of function 'gmtime_s' [-Wimplicit-function-declaration]
     gmtime_s(&ptm, &rawtime);
     ^~~~~~~~

```

### Microsoft Visual Studio
- Install Visual Studio 2017 (Community) witch C/C++ support on Windows 7 64 bit system used in MinGW-example.
Note: The [open62541 website](https://open62541.org/) references "MSVS Windows binaries" but the link does lead to a page where no download can be found.
- Create new empty C++ project; disable precompiled headers.
- Copy the 3 files from MinGW-example to project folder; import files into project.
- Build project (with x64 or x32 configuration).
- Error: :heavy_multiplication_x: 
Error message:
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol __WSAFDIsSet referenced in function ServerNetworkLayerTCP_listen	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_accept referenced in function ServerNetworkLayerTCP_listen	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_bind referenced in function addServerSocket	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_closesocket referenced in function ClientNetworkLayerTCP_close	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_connect referenced in function UA_ClientConnectionTCP	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_ioctlsocket referenced in function socket_set_blocking	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_listen referenced in function addServerSocket	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_recv referenced in function connection_recv	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_select referenced in function UA_ClientConnectionTCP	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_send referenced in function connection_write	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_setsockopt referenced in function ServerNetworkLayerTCP_add	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_shutdown referenced in function ClientNetworkLayerTCP_close	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_socket referenced in function UA_ClientConnectionTCP	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_gethostname referenced in function ServerNetworkLayerTCP_start	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_WSAStartup referenced in function UA_ClientConnectionTCP	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_WSACleanup referenced in function ServerNetworkLayerTCP_stop	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_WSAGetLastError referenced in function UA_ClientConnectionTCP	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_getaddrinfo referenced in function UA_ClientConnectionTCP	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_freeaddrinfo referenced in function UA_ClientConnectionTCP	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp_getnameinfo referenced in function ServerNetworkLayerTCP_add	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\open62541.obj	1	
Error	LNK1120	20 unresolved externals	opcua_libc	C:\Users\D-STF2\Desktop\DEV\opcua_libc\opcua_libc\x64\Debug\opcua_libc.exe	1	
```

### Ubuntu/Linux
- Open WSL Ubuntu terminal.
- Use the files/folder structure from MinGW-example (copy folder `test` to `home`).
- Change into folder `test`.
- Run `gcc -std=c99 open62541.c myClient.c -o myClient`.
- Success :heavy_check_mark: 

Checklist
=========

 - [x] open62541 Version (release number or git tag):
 0.3-rc1
 - [ ] Other OPC UA SDKs used (client or server): 
None
 - [x] Operating system:
Windows 7 64bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
Not necessary; does not compile
 - [ ] Wireshark network dump attached
Not necessary; does not compile
 - [x] Self-contained code example attached
Not necessary; uses only code from:
[https://open62541.org/](https://open62541.org/)
[https://open62541.org/doc/0.3/tutorial_client_firststeps.html](https://open62541.org/doc/0.3/tutorial_client_firststeps.html)
 - [ ] Critical issue
Maybe? Depends on whether Windows support is critical…

lehned, 2018-04-12:
Hi, 
for mingw32 to support _gmtime_s()_ it needs to have _MINGW_HAS_SECURE_API_. Otherwise it must be compiled with build option _--enable-secure-api_ or manual implementation by _gmtime()_.

For linker problems _undefined reference ..._, you need to add to the linker:
* wsock32
* w2_s32

In your case:
 `gcc -std=c99 open62541.c myClient.c -o myClient -lwsock32 -lws2_32`

Best regards!

genericFJS, 2018-04-12:
Thank you for your answer.
I'm not sure what you mean by "it needs to have"? Do you mean as a preprocessor command `-D MINGW_HAS_SECURE_API` (doesn't change anything)? As an additional package to install for mingw32? I tried `gcc -std=c99 open62541.c myClient.c -o myClient.exe -lwsock32 -lws2_32 --enable-secure-api`
This gives me new errors:
```
cc1.exe: error: unknown pass secure-api specified in -fenable
cc1.exe: error: unknown pass secure-api specified in -fenable
```
The command `gcc -std=c99 open62541.c myClient.c -o myClient.exe -lwsock32 -lws2_32` (the additon of `wsock32`) doesn't solve/change the `gmtime_s` error.

Pro, 2018-04-12:
Try to use the mingw64 package, that should work. Mingw32 is missing some features
https://mingw-w64.org/doku.php

Or msys2
https://www.msys2.org/

lehned, 2018-04-12:
Hi, 
> doesn't solve/change the gmtime_s error.
Manual implementation looks like this:
`
errno_t gmtime_s(struct tm* tmi, const time_t* tti)
{
	struct tm *tmi_i = gmtime(tti);
	if (tmi_i == 0)
		return EINVAL;
	memcpy(tmi, tmi_i, sizeof(struct tm));
	return 0;
}
`
Otherwise when your mingw32 has not this feature _MINGW_HAS_SECURE_API_, you can _recompile the compiler_ with option __--enable-secure-api_ or change it as described by _Pro_.
Best regards!

genericFJS, 2018-04-12:
It seems that mingw32 was the culprit. With mingw64 it compiles even without `wsock32`.
Thank you both.

closed 2018-04-12 by genericFJS

genericFJS, 2018-04-12:
Note: Just now I had the idea to look at closed issues.
https://github.com/open62541/open62541/issues/1580 had the same issue: mingw32
There is even the hint on how to build with Visual Studio by adding lws2_32 to the Linker.

effiko, 2018-10-02:
on Windows 10 64 bit - cannot build the sample server
c:\ladico\open62541>gcc --version
gcc (MinGW.org GCC-6.3.0-1) 6.3.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

c:\ladico\open62541>gcc -std=c99 open62541.c myServer.c -lws2_32 -o myServer
open62541.c: In function 'addServerSocket':
open62541.c:38914:19: warning: passing argument 3 of 'setsockopt' makes integer from pointer without a cast [-Wint-conversion]
                   (const char*)&optval, sizeof(optval)) == -1) {
                   ^
In file included from open62541.h:15411:0,
                 from open62541.c:41:
c:\mingw\include\winsock2.h:551:32: note: expected 'int' but argument is of type 'const char *'
 WINSOCK_API_LINKAGE int PASCAL setsockopt(SOCKET,int,int,const char*,int);
                                ^~~~~~~~~~
open62541.c:38914:41: warning: passing argument 4 of 'setsockopt' makes pointer from integer without a cast [-Wint-conversion]
                   (const char*)&optval, sizeof(optval)) == -1) {
                                         ^~~~~~
In file included from open62541.h:15411:0,
                 from open62541.c:41:
c:\mingw\include\winsock2.h:551:32: note: expected 'const char *' but argument is of type 'unsigned int'
 WINSOCK_API_LINKAGE int PASCAL setsockopt(SOCKET,int,int,const char*,int);
                                ^~~~~~~~~~
open62541.c:38913:8: error: too few arguments to function 'setsockopt'
        setsockopt(newsock, IPPROTO_IPV6,
        ^~~~~~~~~~
In file included from open62541.h:15411:0,
                 from open62541.c:41:
c:\mingw\include\winsock2.h:551:32: note: declared here
 WINSOCK_API_LINKAGE int PASCAL setsockopt(SOCKET,int,int,const char*,int);
                                ^~~~~~~~~~
open62541.c: In function 'UA_DateTime_localTimeUtcOffset':
open62541.c:39523:5: warning: implicit declaration of function 'gmtime_s' [-Wimplicit-function-declaration]
     gmtime_s(&ptm, &rawtime);
     ^~~~~~~~