open62541/open62541 #1735: Discussion on Historical Access

peter-rustler-basyskom, 2018-04-12:
Open62541 currently does not support historical access.

We consider this an important feature and would like to discuss how to implement historical access in open62541.

peter-rustler-basyskom, 2018-04-12:
This is the first approach suggested by @jpfr in an E-Mail:

> It would be good if we had a plugin API that allows different storage backends for Historical Data Access (HDA).
There exist a wide range of time-series databases for different use cases. SQL databases with a datetime primary key, column-oriented databases, key-value stores, etc.
A specialized time-series database can split the payload for internal analytics. But the most flexible / general approach would be to just store a bytestring with the binary encoded value payload for every timestamp. 

> A not-too-complicated storage backend would be RocksDB. RocksDB is an ordered key-value store. http://rocksdb.org/
If we use UA_DataTime for the key, we have a barebones in-memory timeseries database with on-disk persistence.
Since the keys are ordered, we can also iterate over the elements in a range (time1, time2).
For this, two methods are defined:

>- Key -> Hash
>- (Key1, Key2) -> Equal / Greater / Smaller

> With a HDA storage backend in place, we need a mechanism to trigger the storage of new entries. We can reuse the existing MonitoredItem implementation for this.
If we allow user-defined callbacks for MonitoredItem notifications, we can notify the local user about changes. And the same mechanism can be used to trigger HDA storage.

> The following are proposed steps towards HDA for discussion.

>- Propose a plugin C-API for exchangeable HDA storage backends
>- Implement an example HDA storage backend (e.g. based on RocksDB)
>- Implement an extension of the MonitoredItem implementation to allow callbacks for new notifications. This includes a new public API for "local" notification callbacks.
>- Propose a C-API for adding local HDA "listeners"/subscribers that write into a HDA backend.
>- Implement the HDA "listeners" based on MonitoredItem Notification callbacks
>- Implement the HDA services (readHistorical / updateHistorical)

basyskom-jvoe, 2018-04-12:
> With a HDA storage backend in place, we need a mechanism to trigger the storage of new entries. We can reuse the existing MonitoredItem implementation for this.
If we allow user-defined callbacks for MonitoredItem notifications, we can notify the local user about changes. And the same mechanism can be used to trigger HDA storage.

If we use the MonitoredItem implementation to trigger the storage of new entries, we would have to sample the node's value. I see two drawbacks of this approach:
- Overhead from periodic sampling, even if the value doesn't change (often)
- Possible missing values in the history if the node is updated with a frequency > sampling interval

There is already a onWrite callback that's called if the node's value is updated. Wouldn't it be better to tap into the mechanism that calls this callback and notify the HDA storage backend if a node is historizing?

jpfr, 2018-04-13:
Hey @basyskom-jvoe,

I see the performance argument as well.
A few points to consider:

- The user himelf might already use the onWrite callback. Can there can be several HDA listeners on the same node (e.g. for different attributes)? So we might have to add a linked list of additional HDA callbacks to every node. I would prefer if we could "contain" the HDA functionality (so that the core representations remain unchanged) and reuse the filter mechanisms from MonitoredItems.
- A user of HDA might expect values with a precise timestamp interval. So we'd have to resort to some kind of time-triggered sampling anyway.
- We had the same discussion for normal MonitoredItems. Our current implementation is simple and "fast enough". Several 10,000 MonitoredItems with 100ms interval on my box on a single core.

There is no definitive answer here. I'm sure you will come up with many more arguments as you dive into the issue. Happy to hear about your thoughts.

jpfr, 2018-04-13:
- If the user wants a notification that is generated without regards to a sampling interval, he can generate an event. The event infrastructure is in the works by @Pro. It also plugs into the MonitoredItems.
- If we don't want to compare attribute values (as we do right no), we could add an int64 counter to every node that is updated when attributes are written. This could give a speed boost for all MonitoredItems with a sampling interval.

peter-rustler-basyskom, 2018-04-13:
Hello,

This is the first proposal for a historical read implementation.
It is provided as a discussion base.

It has not been implemented by using monitored items.
In my opinion this is not a good idea. Normally one want the unmodified source data in the database and using a monitored item for it would alter that data and bring in some fuzzyness.

If for fast moving data you do want to implement some fuzzyness, you can do in the backend.
If you do not want some fuzzyness you stuck with the monitor implementation.
So implementation without monitored item let more freedom to the user of the library.

The first data storage is based on memory. RocksDB do not look like a good solution. It can store key/value pair which would fit for storing data, but i have not found a way to get a range of keys.
So, if you have to get values from timestamp a to timestamp b you have to see every single ms between them if there is a value present. Do I oversee something here ?

Thanks, Peter

PS: Here are some information for the PR

Works:

 * Historical read of multiple notes
 * plugin based memory store backend
 * continuous read
 * range based read (untested)

Does not work:

   * Historical Modify (can be implemented at a later point)
   * Database based storage (can be added with plugin interface)
   * Opt out historical read at compile time (will be done in the near future)

basyskom-jvoe, 2018-04-13:
> The user himelf might already use the onWrite callback. Can there can be several HDA listeners on the same node (e.g. for different attributes)? So we might have to add a linked list of additional [...]

The HistoryRead/HistoryUpdate services for attributes seem to only support the value attribute. At least I didn't see any options to select an attribute in the request structures.
What else would be an use case for having more than one historian **in** the server? There is no way to select a source when doing a history read/update, so there must be one default historian which is queried when doing reads and updates over the network. Other historians could connect to the server over the network and subscribe the necessary nodes they want to keep a history for.

>A user of HDA might expect values with a precise timestamp interval. So we'd have to resort to some kind of time-triggered sampling anyway.

Every write to a node's value attribute has to go through writeValueAttribute(), so this approach wouldn't be missing data points and could use the timestamp from the DataValue when inserting new values into the database. What would be the point of sampling a value with a fixed interval if the value or the timestamps don't change?

peter-rustler-basyskom, 2018-04-13:
Hallo Herr Pfrommer,

ich bin Peter Rustler und arbeite bei der basyskom GmbH als Software-Engineer.

Ich bin im Moment im open62541 Projekt aktiv und arbeite an dem "HistoricalRead" und den "AbsoluteFilter".
Wir hier bei Basyskom würden Sie gerne um ein Telefonat mit dem Thema "Historical Read" bitten.

Wir sind am Montag zwischen 10 und 16 Uhr im Büro.

Haben Sie da Interesse? Wann würde es Ihnen passen?

Vielen Dank, Peter Rustler

jpfr, 2018-04-16:
Hey basykom-crew.
As discussed, let's start with a simple in-memory implementation and add a full database with persistence later on.

With regards to RocksDB. You can iterate over ranges as keys are ordered according to a user-defined relation. You also don't need precise (existing) start and end elements for an iterator. If the start key of an iterator does not exist, the next existing element (according to the key order) is taken for the start.

C++
https://github.com/facebook/rocksdb/blob/af95aecd01b2b0e19c8affb2a2594ac7a846f285/include/rocksdb/iterator.h#L46

C
https://github.com/facebook/rocksdb/blob/af95aecd01b2b0e19c8affb2a2594ac7a846f285/include/rocksdb/c.h#L389

Anyway, adding a full-blown backend database is postponed to a second step.
Which db this shall be is also up for proposals.

jpfr, 2018-04-19:
@merlingeorge555 
Issue #1745 was closed. There was no actual issue description posted.
Please reopen if required.

peter-rustler-basyskom, 2018-05-18:
Hi,
the #1740 is ready for a first review.
Thanks ....

NIPE-SYSTEMS, 2018-05-24:
From a use case with high amount of historical data points:

Consider a source object which produces a lot of data points in a short period of time. The database may get large pretty fast. To counter that, a feature like retention from [Graphite](https://graphiteapp.org/) may help (see [Carbon configuration](https://graphite.readthedocs.io/en/latest/config-carbon.html#storage-schemas-conf)). With retention setup, the database would keep itself small (less administration required). Retention might be non-trivial to implement and it is a corner case of open62541 so maybe it is not that important?

jpfr, 2018-09-01:
The API and a first plugin have been added to master.

closed 2018-09-01 by jpfr