open62541/open62541 #1737: Method call returns seemingly random value

genericFJS, 2018-04-12:
Description
===========
I tried writing a function call according to the given [example](https://github.com/open62541/open62541/blob/master/examples/client.c). I added code from the [tutorial](https://open62541.org/doc/0.3/tutorial_client_firststeps.html) to verify that the basics are working.
I want to call the method `ReadVar` with the argument `/plc/db2001.dbb39`:
[image]
I am using `UA_Client_call(client,
		UA_NODEID_STRING(2, "/Methods"),
		UA_NODEID_STRING(2, "/Methods/ReadVar"),
		1, &input, &outputSize, &output)` (full example below).
I expect the output `4`. But I get two random numbers followed by 9990… for example 439990 or 309000.
Note that the server is working (it works in C# using [open-ua-client](https://github.com/convertersystems/opc-ua-client) and with the UA SDK). The UA SDK example:
[image]

What am I is going wrong?

My code:

```
#include <stdio.h>
#include "open62541/open62541.h"

int main(void) {
	UA_Client *client = UA_Client_new(UA_ClientConfig_default);
	UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://10.1.21.31:4840");
	if (retval != UA_STATUSCODE_GOOD) {
		UA_Client_delete(client);
		return (int)retval;
	}

	UA_Variant value;
	UA_Variant_init(&value);

	/* NodeId of the variable holding the current time */
	const UA_NodeId nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS_CURRENTTIME);
	retval = UA_Client_readValueAttribute(client, nodeId, &value);

	if (retval == UA_STATUSCODE_GOOD && UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_DATETIME])) {
		UA_DateTime raw_date = *(UA_DateTime *)value.data;
		UA_DateTimeStruct dts = UA_DateTime_toStruct(raw_date);
		UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "date is: %u-%u-%u %u:%u:%u.%03u\n",
			dts.day, dts.month, dts.year, dts.hour, dts.min, dts.sec, dts.milliSec);
	}
	/* Clean up */
	UA_Variant_deleteMembers(&value);

	/* Call a remote method */
	UA_Variant input;
	UA_String argString = UA_STRING("/plc/db2001.dbb39");
	UA_Variant_init(&input);
	UA_Variant_setScalarCopy(&input, &argString, &UA_TYPES[UA_TYPES_STRING]);
	size_t outputSize;
	UA_Variant *output;
	retval = UA_Client_call(client,
		UA_NODEID_STRING(2, "/Methods"),
		UA_NODEID_STRING(2, "/Methods/ReadVar"),
		1, &input, &outputSize, &output);
	if (retval == UA_STATUSCODE_GOOD) {
		printf("Method call was successfull, and %lu returned values available.\n",
			(unsigned long)outputSize);
		printf("\tExpected Output: %hhx\n", '4');
		printf("\t         Output: %x\n", output);
		UA_Array_delete(output, outputSize, &UA_TYPES[UA_TYPES_VARIANT]);
	} else {
		printf("Method call was unsuccessfull, and %x returned values available.\n", retval);
	}
	UA_Variant_deleteMembers(&input);

	/*
	Working method request in C# using https://github.com/convertersystems/opc-ua-client:
		new CallMethodRequest {
			ObjectId = NodeId.Parse("ns=2;s=/Methods"),
			MethodId = NodeId.Parse("ns=2;s=/Methods/ReadVar"),
			InputArguments = new Variant[1] { "/plc/db2001.dbb39" }
		}
	*/

	UA_Client_delete(client); /* Disconnects the client internally */
	return UA_STATUSCODE_GOOD;
}
```

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag):
0.3rc1
 - [x] Other OPC UA SDKs used (client or server): 
Sinumerik server; https://github.com/convertersystems/opc-ua-client and UA SDK for verification/testing
 - [x] Operating system:
Windows 7 64bit with MSVS2017
 - [x] Self-contained code example attached
 - [ ] Critical issue

schroeder-, 2018-04-12:
You are printing the pointer Value of output, the value is stored in output->data.

genericFJS, 2018-04-12:
Thank you for your answer. But the problem persists:
If I change `printf("\t         Output: %x\n", output);` to `printf("\t         Output: %x\n", output->data);`, I get a similar problem: It returns two random numbers followed by `78c0` instead of `9990`.

schroeder-, 2018-04-12:
Use:
UA_Int32 value = (UA_Int32) output->data;
printf("\t Output: %x\n", value);

genericFJS, 2018-04-12:
That output is always 1… even when I change the adress. That doesn't seem to do the trick either.
I am confused, that there are changing parts in  `output`/`output->data`. The information on the server is static: always 4. Thus, the output data should never change. Is there something wrong with my method call?

genericFJS, 2018-04-13:
Thanks @Stasik0 but sadly that doesn't help either. 
As stated: `output` and `output[0].data` do not have the same values when I run the programm multiple times: so the problem is not (primarily) how to display the output, but to get a consistent output at all. As @schroeder- stated, `output` is a pointer… but why does `output[0].data` look so similar? It seems to be a pointer, too. How can that be?
Here are some printfs I ran:
```
printf("\tExpected Output: %x\n", '4');
printf("\t     Output int: %x\n", *(UA_Int32*)output[0].data);
printf("\t         Output: %x\n", output);
printf("\t    Output data: %x\n", output[0].data);
```
The output is (where the first two hex values [here `3f`] are changing when I run it again):
```
Expected Output: 34
     Output int: 1
         Output: 3f9880
    Output data: 3fc3c0
```
May it be, that there is something wrong how I start the procedure or prepare the variables (although the lines are copy-pasted from the official examples)?
I tried experimenting and verified, that the input is correct (`/plc/db2001.dbb40` correctly gives an error, since the entry only has 40 bytes till b39; other bytes such as `/plc/db2001.dbb2` give different Int-Outputs [which are wrong, too, though]). Strangely I can change the object node to whatever I want - the call still works (i.e. `UA_NODEID_STRING(2, "Bratwurst")`).
As stated before, I successfully implemented what I want in C# using [open-ua-client](https://github.com/convertersystems/opc-ua-client):
```
var response = await channel.CallAsync(
	new CallRequest {
		MethodsToCall = new[] {
			new CallMethodRequest {
				ObjectId = NodeId.Parse("ns=2;s=/Methods"),
				MethodId = NodeId.Parse("ns=2;s=/Methods/ReadVar"),
				InputArguments = "/plc/db2001.dbb39"
			}
		}
	}
);
CallMethodResult result = response.Results[0];
int resultInt = Int32.Parse(result.OutputArguments[0].ToString()); // is correct: 34
```
How can it be that my open62541 implementation does not work despite the structure of my two implementations seeming almost identical?

schroeder-, 2018-04-13:
output->data or output[0].data is always a pointer where the value is allocated. First what datatype is your return value? For that check output[0].type. For example for Int32 check:
```
if(output[0].type == &UA_TYPES[UA_TYPES_INT32]){
    UA_Int32 value = *(UA_Int32*)output[0].data;
}
```

genericFJS, 2018-04-13:
That's a good idea, I have not thought of that yet.
The datatype seems to be String:
```
printf("\t    Output type: %s\n", output[0].type->typeName);
printf("\t  Output string: %s\n", *(UA_String*)output[0].data);
```
output:
```
  Output type: String
Output string: ☺
```
But that is obviously still far from the String or Integer `4` or the hex value `34`…
It looks as if the method returns "too much": I request a byte (`b39`) and get something that is obviously more than a byte / an ASCII character. Still: as it works flawlessly with open-ua-client and UA SDK, I don't think the fault lies with the server.

schroeder-, 2018-04-13:
UA_String is a struct so you cant print it that way

`UA_String value = *(UA_String*)output[0].data);`
`printf("\t  Output string:  " UA_PRINTF_STRING_FORMAT "\n",  UA_PRINTF_STRING_DATA(value)); `

genericFJS, 2018-04-13:
That did the trick.
For a different argument `/plc/db2001.dbb2` I expect a 65 (int) / A (char). With your hint it works as expected:
```
UA_String strval = *(UA_String*)output[0].data;
printf("\tOutput UAstring: " UA_PRINTF_STRING_FORMAT "\n", UA_PRINTF_STRING_DATA(strval));
long intval = strtol(strval.data, NULL, 10);
printf("\t    Output char: %c\n", intval);
```
output:
```
Output UAstring: "65"
    Output char: A
```
Thank you very much!

closed 2018-04-13 by genericFJS