open62541/open62541 #1741: Error while building open62541.c.obj for Arduino

Corsin91, 2018-04-15:
Hi,
I was trying to to use #1595 for running a simpel UA-Client on my Arduino.  During the build of open62541.c.obj i get following error

` In function "ua_namespace0": `
`error: unable to find a register to spill in class "NO_REGS" `

Used arguments:
UA_ARCHITECTURE=arduino
UA_AMALGAMATION_ARCHITECUTRES=arduino
UA_ENABLE_AMALGAMATION=ON
CMAKE_TOOLCHAIN_FILE=cmake/ArduinoToolchain.cmake
ArudinoToolchain from https://github.com/Pro/arduino-cmake

I am not quite sure if this error is related to open62541-Code or a compiler error, but i thought perhaps someone already had a similar error.

Any ideas how to fix this error? 

Best regards

Corsin91, 2018-04-15:
Thanks Pro, the GNU compiler was too old.
I have installed the Arduino IDE 1.8.5 but, the compiler was 4.9.2. 
I downloaded the latest compiler version from http://blog.zakkemble.co.uk/avr-gcc-builds/ and followed the given steps.
Now i can successfully compile it, but another error occurs in 
` .../avr/bin/ld.exe: region 'text' overflowed by 187214 bytes` when trying to generate the hex-File
I think the *.c and *.h files are too huge.

Have you any good advice how to make it smaller? Trying to use LogLevel Fatal and MinSizeRel in cmake but the two files don't change their size (1706 KB for open62541.c and 1432 KB for open62541.h)
I am quite new in using cmake...

Pro, 2018-04-15:
#1700 should help you getting started

closed 2018-04-15 by Pro